raspberrypi/rpi-imager #577: Move Raspberry Pi OS (64-bit) to main page

theofficialgman, 2023-04-12:
Raspberry Pi OS (32-bit) is still the recommended image and it has the most downloads according to your download statistics.

Raspberry Pi OS (64-bit) is the second most downloaded image and users have to search through the "other" category in order to even find it (by expanding the page and scrolling). This should be changed so that it is found on the main page alongside the 32-bit image or replace the 32-bit image on the main page.

[image]
[image]

0lzi, 2023-06-07:
https://downloads.raspberrypi.org/os_list_imagingutility_v3.json

This is where the change would need to be made looking at the [src/config.h](https://github.com/raspberrypi/rpi-imager/blob/qml/src/config.h)

colinwang1703, 2023-07-07:
> https://downloads.raspberrypi.org/os_list_imagingutility_v3.json
> 
> This is where the change would need to be made looking at the [src/config.h](https://github.com/raspberrypi/rpi-imager/blob/qml/src/config.h)

But it's still better to default.

0lzi, 2023-07-07:
> > https://downloads.raspberrypi.org/os_list_imagingutility_v3.json
> > This is where the change would need to be made looking at the [src/config.h](https://github.com/raspberrypi/rpi-imager/blob/qml/src/config.h)
> 
> But it's still better to default.

Agreed, 64bit should be the default

ghollingworth, 2023-07-07:
Sorry, wrong button, but as I say this is 'in the works'

theofficialgman, 2023-07-08:
I don't see why its taking months to do.
You could have put it on the main page right below the recommended in two mintues with a tiny change to the json (or update more likely your CI that updates the json automatically).

It is criminal for it to be hidden away in the "other" category.

theofficialgman, 2023-10-12:
hoping this can be changed as soon as possible now that bookworm has released.

we would not want the majority of users to install a 32bit image again in their Pi4 and Pi5 when a 64bit image is objectively better for these systems when configured with enough ram (4GB+) to take advantage of 64bit's extended ram address space.

lurch, 2023-10-12:
It's in the works, see #645

tdewey-rpi, 2023-10-19:
Bookworm 64bit is now shown at the top of the hierarchy for platforms that can run it. 

[image]

closed 2023-10-19 by tdewey-rpi

theofficialgman, 2023-10-20:
@tdewey-rpi I'll give you the benefit of the doubt that these issues were not intended and did not happen in testing but right now pretty much everything is very broken https://github.com/raspberrypi/rpi-imager/issues/577