raspberrypi/rpi-imager #579: Will rpi-imager save the firmware written to the sd card?

Kay-Anonymous, 2023-04-14:
Hello, gurus.
Will rpi-imager save the firmware written to the sd card?
If yes, where are these images stored?

Kay-Anonymous, 2023-04-15:
I installed version 1.7.3 on ubuntu20.04, and I did not find the lastdownload.cache file. But it looks like there is plenty of disk space

maxnet, 2023-04-15:
Does exists for me under (K)Ubuntu:

```
max@lynx:~$ ls -l ~/.cache/Raspberry\ Pi/Imager/lastdownload.cache 
-rw-rw-r-- 1 max max 795265696 Apr  9 18:35 '/home/max/.cache/Raspberry Pi/Imager/lastdownload.cache'
```

Kay-Anonymous, 2023-04-17:
Well, maybe my previous writes are not cached, I will download a copy directly, thank you ～
```bash
$ ll ~/.cache/Raspberry\ Pi/Imager/
oslistcache0/ oslistcache1/ qmlcache/  
```

closed 2023-04-17 by Kay-Anonymous