raspberrypi/rpi-imager #580: Latest release not linked on main web page

dsl101, 2023-05-03:
I know this isn't necessarily an issue with the imager itself, but for ages I was stuck because the download link [here](https://www.raspberrypi.com/software/) which is `https://downloads.raspberrypi.org/imager/imager_latest.exe` is pointing to version 1.6.2, which is missing some significant functionality (like the customisation options). Eventually I found this repo and the releases, and got 1.7.4.

Note the other platform links are also outdated—1.7 for MacOS and 1.6.1 for Ubuntu.

XECDesign, 2023-05-04:
Many thanks for spotting and reporting this. Looks like the script that generates the redirects doesn't sort the files properly.

I've fixed the links manually for now and will take a look at making sure they get generated properly next time.

XECDesign, 2023-05-04:
Should be sorted so that it doesn't happen again.

closed 2023-05-22 by maxnet