raspberrypi/rpi-imager #583: Headless wpa_supplicant.conf configuration stopped working

eximius313, 2023-05-16:
With the newest Imager (1.7.4) when I tick "Configure wireless LAN" checkbox and just run the card - the wifi is configured properly and wlan0 interface is present in `ifconfig`.
But when I copy my `wpa_supplicant.conf` file (which contains additional networks) to the root folder for SD card (like I always did for headless configuration with older version of Imager) - the file is indeed copied into `/etc/wpa_supplicant/` but wifi is not connected and `wlan0` is **not** present in the `ifconfig`.
It is the same file and the same routine I always did with previous versions of Imager.
What is surprisingL when I then run `raspi-config` and add any dummy network `Foo` with password `12345678` - it is added to my file, `wlan0` is up and wifi is connected to my real network.

Could you please fix that?

XECDesign, 2023-05-16:
Are you saying the behaviour was different with 1.7.3 and upgrading to 1.7.4 is what breaks it, or just that you're seeing the problem with 1.7.4? It could be a raspberrypi-sys-mods or raspberrypi-net-mods issue rather than rpi-imager.

eximius313, 2023-05-16:
I'm saying that the problem is after upgrading to `1.7.4` but I don't remember what previous version I had (could be from 6 months ago)

maxnet, 2023-05-16:
And you are selecting the same RPI OS .img file you downloaded previously manually with "use custom" ?
Or did you install the latest RPI OS version today, which is likely a different RPI OS version than 6 months ago?

If so, it doesn't sound like an Imager issue to me...

eximius313, 2023-05-16:
It was the latter ;(
Ok, how can I pinpoint the issue?
Shall I download old Imager and use newest image?
Or try to find old image and use it with newest Imager?

eximius313, 2023-06-04:
Ok, I've installed newest Imager (v1.7.5) and verified, that image from [2022-04-07](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/) works, but later images - for example [2022-09-07](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-07/) do not.

Where shall I raise the issue?

closed 2023-06-04 by eximius313

XECDesign, 2023-06-05:
https://github.com/RPi-Distro/repo/issues

eximius313, 2023-06-21:
@XECDesign I've created the issue here: https://github.com/RPi-Distro/repo/issues/328
Can I help with investigation the problem in any way?