raspberrypi/rpi-imager #584: System Error wlanapi.dll was not found

rgl, 2023-05-19:
When running in Windows Server 2022, the following error prevents the application from starting:

[image]

```
---------------------------
rpi-imager.exe - System Error
---------------------------
The code execution cannot proceed because Wlanapi.dll was not found. Reinstalling the program may fix this problem. 
```

This happens because, on Windows Server, the Wireless LAN Service is not not installed by default.

This left me a wondering... why does rpi-imager require that service?

Can that dependency be dropped?

rgl, 2023-05-19:
I did not trim it, I did not even noticed that Windows Server didn't install WLAN services until now (I never tried to use WLAN in Windows Server).

I'm using the Windows Server 2022 Evaluation ISO (FWIW, I'm using the stuff at https://github.com/rgl/windows-vagrant to create a base image and burn it into a physical machine).

maxnet, 2023-05-21:
Ok, wasn't aware it wasn't present by default on Windows Server.
The developer documentation only mentioned a special redistributable needed for Windows XP prior to SP3.

https://learn.microsoft.com/en-us/windows/win32/api/wlanapi/nf-wlanapi-wlanopenhandle

```
Requirements
 	 
Minimum supported client	Windows Vista, Windows XP with SP3 [desktop apps only]
Minimum supported server	Windows Server 2008 [desktop apps only]
Target Platform	Windows
Header	wlanapi.h (include Wlanapi.h)
Library	Wlanapi.lib
DLL	Wlanapi.dll
Redistributable	Wireless LAN API for Windows XP with SP2
```

Anyway.
Can you try if this build works better for you: [imager-1.7.5.exe](https://www.maxnet.eu/tmp/175-wlanapi/imager-1.7.5.exe)

rgl, 2023-05-22:
I think they failed to mention that although no WLAN redists are needed, in the Server editions, the `Wireless LAN Service` feature must be installed.

I've tested imager-1.7.5.exe and it works nicely. :-)

maxnet, 2023-05-22:
Fixed in 1.7.5

closed 2023-05-22 by maxnet

Hu-Wentao, 2024-07-05:
> I think they failed to mention that although no WLAN redists are needed, in the Server editions, the `Wireless LAN Service` feature must be installed.
> 
> I've tested imager-1.7.5.exe and it works nicely. :-)

it work for me