raspberrypi/rpi-imager #585: Semantic Version? Version tag format syntax?

rgl, 2023-05-19:
Is this project following [semantic versioning](https://semver.org/)?

Or can it follow its version syntax? Abbreviated version syntax:

```
<valid semver> ::= <version core>
                 | <version core> "-" <pre-release>
                 | <version core> "+" <build>
                 | <version core> "-" <pre-release> "+" <build>
<version core> ::= <major> "." <minor> "." <patch>
```

I'm asking this because tools like [renovate](https://github.com/renovatebot/renovate), out-of-the-box, do not handle versions that do not have 3 digits (e.g. [v1.7.4](https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7.4)), and the latest rpi-imager has four digits (e.g. [v1.7.4.1](https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7.4.1)).

maxnet, 2023-05-21:
Not sure if there are any RPI best practices on version numbering...

But do note that having up to 4 digits has been the standard for Windows desktop applications for a very long time.
https://learn.microsoft.com/en-us/windows/win32/menurc/versioninfo-resource
So to me it sounds a bit strange that Renovate is hard-coded to 3 digits, unless they are only focussing on some web application platform that has that as standard...

rgl, 2023-05-21:
Renovate does [support using other formats](https://docs.renovatebot.com/modules/versioning/). I was trying to suggest using a simpler version number syntax, but I'm good either way, and will keep using the rpi-imager :-)

iyanmv, 2023-05-22:
For me it was also a bit arbitrary to see the latest release as 1.7.4.1 instead of 1.7.5 (it's the first tagged version with 4 digits), but also previous "major" updates were missing the "minor" part (e.g. 1.7 instead of 1.7.0), so I guess the versioning of RPI is less rigid than other projects.

tdewey-rpi, 2023-10-19:
Going forward, I expect to keep the versioning to the SemVer standard. Thanks for the pointer.

closed 2023-10-19 by tdewey-rpi