raspberrypi/rpi-imager #586: Unsupported ZIP Extraction Error

Gramaz, 2023-05-20:
On 1.7.4.1, I cannot extract officially downloaded image files or use the "custom" file either - every time it results in an error in extracting.

maxnet, 2023-05-20:
What operating system are you running Imager on?

The libarchive version that handles uncompression did get updated on  1.7.4.1 if using Windows or MacOS.
(On Linux we use the libarchive that comes with your Linux distribution, so version used may vary).
None of the alpha testers reported this problem, though.

And what exact image file are you using?
(You mention .zip and the official images. But the recent RPI OS images are .xz, not .zip.)

maxnet, 2023-05-22:
>Same issue here.

While writing which OS image?

maxnet, 2023-05-22:
Can you test if this build solves the problem?

[imager-1.7.5-zipfix.exe](https://www.maxnet.eu/tmp/175-zipfix/imager-1.7.5.exe)

ray1025, 2023-05-22:
> Can you test if this build solves the problem?
> 
> [imager-1.7.5-zipfix.exe](https://www.maxnet.eu/tmp/175-zipfix/imager-1.7.5.exe)

Yes,it's solves the problem.

digitalsleuth, 2023-05-22:
Same issue here, the 1.7.5 zipfix resolved the issue for me as well.

maxnet, 2023-05-22:
Fixed in 1.7.5

closed 2023-05-22 by maxnet

FreeDMR-Digital-Network, 2023-09-13:
for me the 1.7.5 does not work, still zip error (Win7, zip, error: 9 deflation 64bit

maxnet, 2023-09-13:
> for me the 1.7.5 does not work, still zip error (Win7, zip, error: 9 deflation 64bit

Proprietary zip flavors are indeed not supported.

https://www.zlib.net/zlib_faq.html#faq40