raspberrypi/rpi-imager #587: rpi-imager deb file won't install on Ubuntu  22.04.2

ichibrosan, 2023-05-22:
imager for linux won't install on Ubuntu 22.04.2 LTS
```
root@z9m9z:/home/doug/Downloads# dpkg -i imager_1.7.4_amd64.deb
Selecting previously unselected package rpi-imager.
(Reading database ... 206974 files and directories currently installed.)
Preparing to unpack imager_1.7.4_amd64.deb ...
Unpacking rpi-imager (1.7.4) ...
dpkg: dependency problems prevent configuration of rpi-imager:
rpi-imager depends on libqt5core5a (>= 5.12.2); however:
Package libqt5core5a is not installed.
rpi-imager depends on libqt5dbus5 (>= 5.0.2); however:
Package libqt5dbus5 is not installed.
rpi-imager depends on libqt5gui5 (>= 5.10.0) | libqt5gui5-gles (>= 5.10.0); however:
Package libqt5gui5 is not installed.
Package libqt5gui5-gles is not installed.
rpi-imager depends on libqt5network5 (>= 5.12.2); however:
Package libqt5network5 is not installed.
rpi-imager depends on libqt5qml5 (>= 5.10.0); however:
Package libqt5qml5 is not installed.
rpi-imager depends on libqt5widgets5 (>= 5.0.2); however:
Package libqt5widgets5 is not installed.
rpi-imager depends on qml-module-qtquick2; however:
Package qml-module-qtquick2 is not installed.
rpi-imager depends on qml-module-qtquick-controls2; however:
Package qml-module-qtquick-controls2 is not installed.
rpi-imager depends on qml-module-qtquick-layouts; however:
Package qml-module-qtquick-layouts is not installed.
rpi-imager depends on qml-module-qtquick-templates2; however:
Package qml-module-qtquick-templates2 is not installed.
rpi-imager depends on qml-module-qtquick-window2; however:
Package qml-module-qtquick-window2 is not installed.
rpi-imager depends on qml-module-qtgraphicaleffects; however:
Package qml-module-qtgraphicaleffects is not installed.

dpkg: error processing package rpi-imager (--install):
dependency problems - leaving unconfigured
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu3) ...
Processing triggers for desktop-file-utils (0.26-1ubuntu3) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for man-db (2.10.2-1) ...
Errors were encountered while processing:
rpi-imager
root@z9m9z:/home/doug/Downloads#

```
Regards,

Douglas Goodall

maxnet, 2023-05-22:
dpkg is the wrong command if you would like dependencies to be auto-installed.
Use `apt` instead.

```
sudo apt install ./*.deb
```

closed 2023-05-22 by maxnet

ichibrosan, 2023-05-23:
Well thank you for telling me that. Someone showed me dpkg years ago and I have been running around sure of myself since. Shucks.  Thanks again.  :-)Sent from my iPhone

Closed #587 as completed.