raspberrypi/rpi-imager #589: Creating an image break the SD card

alexfauquette, 2023-06-03:
I have a new SD card, but when trying to set up an OS, it does the writing. Then fail with 

> Error reading from storage.
> SD card may be broken.

I can reformat the SD card. After that, it's still working I can write/read files on it. If I try again the installation, if fails again

ghollingworth, 2023-06-04:
1) Check you have the latest version of rpi-imager

https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7.5

2) Use a different USB SD card reader...  If it is an inbuilt SD card reader in the PC then that often has trouble because Windows drivers don't work well with them.

closed 2023-06-04 by ghollingworth

lurch, 2023-06-04:
Another possibility (after you've checked the above two things) would it being a "fake" SD card, where it's lying about its true capacity (e.g. it might advertise itself as an 8GB card, and any writes in the first 2GB work fine, but any writes above 3GB get lost).

alexfauquette, 2023-06-04:
Thanks for your answers

Same error with v1.7.5. It's a built-in SD card reader but I'm on Ubuntu

I don't think it can be a fake storage. It's a card I just bought. A [Philips FM64MP45B](https://www.philips.fr/c-p/FM64MP45B_00/cartes-micro-sd)

ghollingworth, 2023-06-04:
Does it work correctly with dd?

maxnet, 2023-06-04:
After it happens you also may want to run "dmesg" on a terminal and look at the last number of lines.
If the kernel mentions any IO errors there, then it is not something we can fix.

alexfauquette, 2023-06-07:
Does it work correctly with dd?

Yes. I've used "Image disc writer" to write the content of `2023-05-03-raspios-bullseye-armhf.img.xz`, and now the Raspberry is working 🎉

Thanks for the support :)

maxnet, 2023-06-07:
Problem with "dd" and a lot of other tools is that they do not verify if the write was successful.
Note that Imager reported a problem READING, not a problem writing.

alexfauquette, 2023-06-08:
Yes, but after his writing the SD card was unusable, and I had to reformat it to get access back to it

lurch, 2023-06-08:
[Etcher](https://etcher.balena.io/) is another disk-image-writing tool that also actually does verification, so it'd be interesting to hear if that has similar problems to RPi Imager?

maxnet, 2023-06-08:
>Yes, but after his writing the SD card was unusable, and I had to reformat it to get access back to it

That is because we do not write the partition table, if verification fails.

alexfauquette, 2023-06-26:
```bash
$ sudo dd if=/dev/mmcblk0 bs=4MiB count=$((4143972352/(4*1024*1024)+1)) of=readback.img

dd: error reading '/dev/mmcblk0': Input/output error
1+1 records in
1+1 records out
4325376 bytes (4,3 MB, 4,1 MiB) copied, 1,26978 s, 3,4 MB/s
```

Since it returned an error, I decided to try with another address (`mmcblk0p1` instead of `mmcblk0`). But it returned the same error

[image]

```
$ sudo dd if=/dev/mmcblk0p1 bs=4MiB count=$((4143972352/(4*1024*1024)+1)) of=readback.img

dd: warning: partial read (1306624 bytes); suggest iflag=fullblock

dd: error reading '/dev/mmcblk0p1': Input/output error
1+45 records in
1+45 records out
9895936 bytes (9,9 MB, 9,4 MiB) copied, 117,695 s, 84,1 kB/s
```

Then Here is the result of the last steps

```bash
$ sudo truncate -s 4143972352 readback.img
$ md5sum readback.img

e8144f09064178168094581c8242492c  readback.img
```