raspberrypi/rpi-imager #592: [OS Request] Alpine Linux

0lzi, 2023-06-07:
Would love to see in the Other general-purpose OS section , an option for Alpine linux.

[image]

https://alpinelinux.org/downloads/

ghollingworth, 2023-06-07:
Whoever is responsible for creating the images would have to provide a suitable .json file for RPi-Imager to read

rootwyrm, 2023-06-08:
@ghollingworth can you please point me in the direction of the documentation for the image customization pieces?
I would be happy to add this to my [alpine-imager](https://github.com/rootwyrm/alpine-imager) repository, which is already partly designed for this exact situation (where the image needs customized.)
The way I do it is very different from the Alpine images because of the way Alpine handles imaging, which is pretty unfriendly to customization. I use their signed minimal rootfs, verify, add official packages and some customization hooks (not all fully implemented yet.) Finishing and extending those hooks to work with the imager should be pretty trivial.

maxnet, 2023-06-08:
>@ghollingworth can you please point me in the direction of the documentation for the image customization pieces?

- Add "init_format": "cloudinit" to json.
- Parse the user-data file on FAT partition. Format: https://cloudinit.readthedocs.io/en/latest/index.html

rootwyrm, 2023-06-09:
@maxnet other than 'boots, runs, and has networking' are there any specific must-have or must-not-have requirements for images? I have begun testing here and things are looking somewhat promising.

Unfortunately, as I (like many people) don't have any spare RPi4's for testing, I will need some assistance validating there since there have been some overlay problems in the past.

rootwyrm, 2023-06-10:
And I have success with internal development, and true to Alpine form, just 125MB gzipped. But some additional fixes will be needed as Alpine runs a newer (and stricter) cloud-init.

- `meta-data` must be generated, which rpi-imager does not currently do (will try to work on that this weekend; this will not cause problems with less-strict ones)
- [this also applies to Alpine](https://github.com/raspberrypi/rpi-imager/blob/b49408781a3c347bd6f6c057c68bb34d6c06ad10/src/OptionsPopup.qml#L751-L753), but to a more limited extent. I need to do further investigation.
- Alpine has switched from `sudo` to `doas` and provides a patched `cloud-init` that understands `doas`; I need to check on whether `sudo` will remain available in the future.

@maxnet @ghollingworth one thought I had was adding `"init_format": "alpine_cloudinit"`  in order to avoid potential collisions, especially as I would like to extend Alpine to support ethernet + wifi. Does this sound agreeable? Or would it make more sense to perhaps extend with something like `"distribution": "alpine"`?

maxnet, 2023-06-11:
>[this also applies to Alpine](https://github.com/raspberrypi/rpi-imager/blob/b49408781a3c347bd6f6c057c68bb34d6c06ad10/src/OptionsPopup.qml#L751-L753), but to a more limited extent. I need to do further investigation.

Just make sure your distribution your distribution supports the "regulatory-domain" setting in network config instead.
As that is what we likely will be using once Ubuntu retires Ubuntu 20 (that does not support it, and reacts badly when it does is present).
(Had a commit that added support for that: https://github.com/raspberrypi/rpi-imager/commit/31252bf1c4af5c375fc7588855f7da8ca4b6a585 but reverted it because of this issue)

>@maxnet @ghollingworth one thought I had was adding "init_format": "alpine_cloudinit" 

I am not a big fan of that, as then things will not work properly if people use an .img they downloaded manually, instead of from the repository.

ghollingworth, 2023-06-11:
@XECDesign has been working on a different way of doing this, by implementing a TOML config file which is then processed by the image on first boot (in an initramfs), rather than using a script.  That way we can keep the specification of the config file constant while changing the implementation within the image (i.e. you'll not have to make sure Imager is updated in line with the image)

But it requires a bunch of other things working properly before we get there...  So for the moment, I'd suggest sticking with whatever works until that is available.

rootwyrm, 2023-06-15:
Unfortunately haven't had much time to work on this, and unfamiliar with Qt5, so haven't had luck getting it to build on Alma 8.8. (Instructions refer to packages that aren't there any more.)

> Just make sure your distribution supports the "regulatory-domain" setting in network-config instead.

Alpine is unfortunately both easier and more complicated, since it uses the `crda` method. However, adding an `/etc/modprobe.d/cfg80211.conf` on other distributions with the exact same setting wouldn't cause any harm. 

Adding the `doas` section WILL break cloud-init on other systems, and I definitely agree that adding a whole other cloudinit setup that's specific to Alpine is not great. I'll need to check what the intent is with `sudo` removal though, since technically, it's deprecated as of 3.17. 

> @XECDesign has been working on a different way of doing this, by implementing a TOML config file which is then processed by the image on first boot (in an initramfs), rather than using a script. [...]

This definitely would be the 'perfect' solution in my view as well. Especially if it can key around the JSON or similar to select the distribution pattern. Alpine almost definitely will continue to require tweaks, but at a lower intensity. Breaking changes are a fairly frequent thing between majors (i.e. sudo -> doas, default Python versions, etc.) And they do majors pretty frequently. But if they can be generally covered in the TOML with JSON, that's very manageable.

I think for now what makes the most sense is sticking to minimum-viable that doesn't break anything else, so that would be the crda portion and adding meta-data generation. 

As I mentioned, I'm unfamiliar with Qt5 and QML. So any guidance would be appreciated there. Does this look correct to add the `meta-data` portion? 
https://github.com/raspberrypi/rpi-imager/compare/qml...rootwyrm:rpi-imager:dev-alpine

maxnet, 2023-06-15:
>Unfortunately haven't had much time to work on this, and unfamiliar with Qt5, so haven't had luck getting it to build 
>on Alma 8.8. (Instructions refer to packages that aren't there any more.)

Well, if you want help with that, it may be worth mentioning what packages you do have...
Imager should also compile fine against Qt 6 instead of Qt 5, if you do have packages for that.
(Although that has only been tested on Mac OS X, as it does not bring any advantages on Linux distributions that do have Qt 5)

maxnet, 2023-06-15:
>Does this look correct to add the meta-data portion?

I suggest you double check first if you actually need changes in Imager for a dynamic meta-data potion...

Ubuntu just bundles a static "meta-data" file:

```
# This is the meta-data configuration file for cloud-init. Please refer to the
# cloud-init documentation for more information:
#
# https://cloudinit.readthedocs.io/

# Set the datasource mode to "local". This ensures that user-data is acted upon
# prior to bringing up the network (because everything about the datasource is
# assumed to be local). If you wish to use an HTTP datasource instead, you can
# change this to "net" or override it on the kernel cmdline (see README).
dsmode: local

# Specifies the "unique" identifier of the instance. Typically in cloud-init
# this is generated by the owning cloud and is actually unique (to some
# degree). Here our data-source is local, so this is just a fixed string.
# Warning: changing this will cause cloud-init to assume it is running on a
# "new" instance, and to go through first time setup again (the value is
# compared to a cached copy).
instance_id: cloud-image
```

Think when "dsmode: local" it should consult the user-data file for the hostname.
If it does not do that on Alma that may be a Alma bug that should be fixed.

rootwyrm, 2023-06-15:
> Well, if you want help with that, it may be worth mentioning what packages you do have...

What I found is that for some reason, many distros (Fedora, CentOS, OEL, Alma) no longer have `qt5-qtquickcontrols2-devel` or anything that provides the headers. That should be the only thing actually missing, but obviously, it's a rather big thing to be missing.
This is obviously distribution weirdness, but I've found no explanation or reasoning for why the header package is no longer being built. 

> I suggest you double check first if you actually need changes in Imager to generate a dynamic meta-data file...
> Ubuntu just bundles a static "meta-data" file:

Which is actually non-standard and non-default. Alpine actually follows things the right way; Ubuntu 150% does not and just goes off and does whatever it wants, in part because the new installer relies on non-standard cloud-init.
A `meta-data` is required to be located with the `user-data` when using a valid datasource i.e. NoCloud or configuration drive on a standard (as distributed) cloud-init. The way Ubuntu ships is nowhere near the default configuration of cloud-init. But it's important to note that Ubuntu's setup will work with user-data and meta-data which _is_ designed for cloud-init's defaults. So it won't actually change behavior meaningfully there either. 
So hewing to cloud-init defaults instead of Ubuntu's modified setup will solve for a lot more than one (RHEL-based, Alpine, Slackware, FreeBSD, etc.) by having only an additional NoCloud datasource in `/etc/cloud/cloud.cfg.d/nn_dsnocloud.cfg` (`/usr/local/etc/cloud.cfg.d/` in FreeBSD IIRC, but I can look at it later) in the distribution image and will not require any changes to Ubuntu images.

maxnet, 2023-06-16:
>What I found is that for some reason, many distros (Fedora, CentOS, OEL, Alma) no longer have qt5-qtquickcontrols2-
>devel or anything that provides the headers.

May want to try if it works with just the package counterpart without "-devel".
Think we are only using the QuickControls from QML, which may not need the C++ headers.

>A meta-data is required to be located with the user-data when using a valid datasource i.e. NoCloud or configuration 
>drive on a standard (as distributed) cloud-init.

Note that the information typically provided in meta-data like hostname and instance-id can also be provided to cloud-init through other ways like in cmdline.txt ( ds=nocloud;s=file://boot/user-data;h=myhostname;i=myinstance   ).
Not sure if cloud-init's insistence of there always being a meta-data file even in cases it doesn't add anything is actually correct behavior.

>So hewing to cloud-init defaults instead of Ubuntu's modified setup will solve for a lot more than one (RHEL-based, 
>Alpine, Slackware, FreeBSD, etc.) by having only an additional NoCloud datasource in 
>/etc/cloud/cloud.cfg.d/nn_dsnocloud.cfg

Whether you need to add one or two additional files, it is differing from defaults in both cases.

macmpi, 2023-10-31:
FWI, there's some ongoing work within Alpine build system to generate proper `img.gz` file and `.json`, likely for upcoming (soon) 3.19 release.
Will send the subitem `.json` here for inclusion when ready.

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi

macmpi, 2024-10-22:
@tdewey-rpi @ghollingworth form has been submitted on Sept 27th for Alpine inclusion as supported OS.
It does not show up yet: any hint on roll-out date into [master json](https://downloads.raspberrypi.org/os_list_imagingutility_v4.json)? 
Thanks.