raspberrypi/rpi-imager #593: First write always fails verification

joseph-mccarthy, 2023-06-07:
When writing an image to SD card the first attempt fails image verification. On clicking write again with no settings changed the verification passes. 

OS: Fedora Linux 37 (Workstation Edition)
Imager Version: 1.7.4
Attempting to write: Ubuntu Server 22.04.2 LTS (64 Bit)

Please let me know if you need any more details, or how I could dump debug logs.

maxnet, 2023-06-07:
How reproducible is the issue, as in have you tried it more than a couple times?

And you do are using a natively compiled Imager (either from source or .rpm) and not something container based (flatpack, snap, etc.)?

joseph-mccarthy, 2023-06-07:
It's very repeatable. Insert SD card into SD card reader. Run the imager, it fails verification, and then the second time it always works. Using native .rpm.

maxnet, 2023-06-08:
Unable to reproduce on Fedora 38.
Do note that your Imager and Fedora versions are old.
If you have a spare USB stick, you may want to try writing Fedora 38 to that, booting it, and see if the issue also exists there.
Just running it "live" (without installing to hard disk), and typing "sudo yum install -y rpi-imager && rpi-imager" is fine for testing.
May also want to try a different SD card reader if you have one, just to see if it makes any difference.

[image]

joseph-mccarthy, 2023-06-08:
Thanks I'll try your suggestions and report back.

joseph-mccarthy, 2023-06-09:
I've gone with your suggestions of an upgrade and everything appears to be working correctly. Thanks for your support.

closed 2023-06-09 by joseph-mccarthy