raspberrypi/rpi-imager #594: Blue screen on Win10 after flasing SD card

janoh3, 2023-06-11:
Hello, this is my first report, I hope I do it correctly. I tried flashing image to SD card, it works well until SD card is flashed. Then it makes blue screen. I tried it with two different SD cards several times. The image fortunately works well, but the blue screen should not appear. Older version worked without blue screen, the newest v1.7.5 started to do this. Thanx for fixing.

maxnet, 2023-06-11:
Have you verified that it works again after installing an older version from the releases page?

Note that blue screens are more typical driver and other kernel issues.
Which can come and go away after each automatic Windows update...

Have a photo of the blue screen?
Does it concern a laptop with built-in SD card reader (those often have non-standard driverd)?

colinwang1703, 2023-07-07:
Is it 1.7.5 yet? Anyway, I still prefer to use the old version.

bsimmo, 2023-09-20:
We have a suite of computers (modernish desktop, 1650Ti era.
They often bluescreen with 1.7.5 (I just posted on the forum, made the lesson a bit messy).
I moved to 1.7.4.1 and there have been no problems.

So something in 1.7.5 isn't working anymore.

It would either Bluescreen at the start of Write or the end of Verify.
I don't know if it happened in between as they rebooted and tried again and again.
Our computers use an image that it loads each boot time so is always the same no matter what the students have done to it.

ndabas, 2023-09-20:
It's basically impossible for anything in user-mode code to trigger a blue screen on Windows. So I'd say it's not anything 

If you are consistently getting a blue screen on a specific action, it's a problem with some driver, most likely -- like the wudfrd.sys that you noticed. It's possible to look at the blue screen info in the system log to identify the driver -- usually the file that caused the fault will be identified in the log. Else there are ways to analyze the  dump/minidump file produced during the BSoD to get to the culprit.

The problem with wudfrd.sys is that it is actually a wrapper/loader for user-mode drivers, so the actual culprit would be something else.

bsimmo, 2023-09-20:
Yes, but only 1.7.5 is causing the problem, 1.7.4.1 does not have the problem (for us). Something in it, in its compilation has changed and is triggering the 'driver/windows' to blue screen.   The program _is the only thing_ that has changed.

I'm not the only one with 1.7.5 problems.
I'm also not saying wudfrd is always the error (or not)

I'm not new to this, A long time ago in the early days of Pi Imager, Realtek drivers on Dell laptops had a problem, that long since gone.  I can write happily on this laptop now.

maxnet, 2023-09-20:
Suggest you double check that you can reproduce it with a normal USB SD card reader first.
For all we know the user-mode driver thingy could be for your internal SD card reader (if it does not speak standard USB MSD, but something custom).

bsimmo, 2023-09-22:
> Suggest you double check that you can reproduce it with a normal USB SD card reader first. For all we know the user-mode driver thingy could be for your internal SD card reader (if it does not speak standard USB MSD, but something custom).

If that's at me, then I can try and find out more details, pretty sure other blue screens came up over the room _wudfrd_ could be a red herring.
It's not a big problem for me, we just changed back to 1.7.4* and got on with it.
I just needed to remember that if they try again (due to the nature, like I said they are all reset to a base image on reboot) it's not installed when they logon.
If I get time and the room is spare, I'll dig around if really needed.
From me it's just feedback to say something altered in 1.7.5 to trigger this.

I'm not even saying it wouldn't go away if we updated and moved to Win11 or I went looking for other fixes.
Just logging it.

maxnet, 2023-09-22:
>It's not a big problem for me, we just changed back to 1.7.4* and got on with it.

There is zero difference in the write code between 1.7.4.1 and 1.7.5
Only in the code that borrows the WLAN password from your system if you use advanced settings.

bsimmo, 2025-12-12:
> There's been enough churn around how we interact with Windows in 2.0 that
> this must be marked stale.
>
> If you're seeing BSoDs with 2.0, I'd be most curious.

For the info, V2 is working fine on the same computers. Though we also
moved to Windows 11 recently too.
No doubt some driver problem.