raspberrypi/rpi-imager #596: advanced menu for custom ssh, wpa and username/password config not working

Jameswill-Winsock, 2023-06-14:
Hello there, so this is my first post here (and ever on GitHub), kindly excuse any errors i make

so while using the imager for raspberry pi os (32-bit, for the zero w), the imager does not create any wpa config or ssh config files, and neither any ssh settings defined in the advanced menu. Everything (wpa_supplement.conf, userconf.txt, blank ssh file) had to be in the end created manually in bootfs, as it wasn't completely headless and i did not have access to a secondary monitor only having my laptop (Linux mint 21 with the imager downloaded from the official website).

This could be a bit of an issue, as it does not make the raspberry pi headless "out of the box", and newer users could end up being annoyed and irritated.

Thanks in advance 😃 👍

maxnet, 2023-06-14:
>so while using the imager for raspberry pi os (32-bit, for the zero w), the imager does not create any wpa config or ssh config files, and neither any ssh settings defined in the advanced menu.

You should see a file called firstrun.sh on the FAT partition instead, which sets all those things.

>Everything (wpa_supplement.conf, userconf.txt, blank ssh file) had to be in the end created manually in bootfs

Those files are not needed if using firstrun.sh
So the absence of those files is not a good indication if Imager does its job correctly.

Jameswill-Winsock, 2023-06-15:
> You should see a file called firstrun.sh on the FAT partition instead, which sets all those things. After booting for the first time, it self destructs, so you can only see it after the write, before you stick the card in the Pi.

Ah, i see...

> Those files are not needed on the FAT partition if using firstrun.sh So the absence of those files is not an indication Imager fails to do its job.

The problem here i have seen is that even with repeated flashing it does not run firstrun.sh for me for some reason (as no user or ssh settings are being created). Is there a way i can check if the pi runs the file via some command?

Jameswill-Winsock, 2023-06-15:
cool, I'll see it. Thanks for the help. I'll check it after i get home from school tommrow, and hopefully close it first thing.

closed 2023-06-18 by Jameswill-Winsock