raspberrypi/rpi-imager #597: What am I doing wrong? Raspi-Imager not saving settings, not configuring wifi properly.

DdraigX, 2023-06-20:
Howdy,

Could use some insight here.

I am currently using 1.7.5 version.  It is not saving the settings properly to the pi.

I started with 1.7.4 and the settings were not saving. Updated to 1.7.5
Then total uninstall including registry keys. 
I've tried running as administrator. 

It is saving some type of hashed value to the pi login/user id. 
Saving some type of hash to the wifi settings.
As far as I can tell not enabling ssh? 

After a total uninstall and removing the registry entry for the settings I double checked to see if firstrun.sh was there.

firstrun.sh was there it said enable ssh, and had the wifi information however the pw was hashed again. Right now I'm sitting here staring at the screen and no wifi is configured. 

What can I check or do to get this fixed?

maxnet, 2023-06-20:
Where did the wifi settings come from?
As in: did it copy them automatically from the computer you are running Imager on, or did you enter them manually?

If manually, be aware that not only the the password but also the SSID is CaSe SenSitiVe.
And if you made any typo in the SSID, you also need to enter the password again, as the hash is a combination of SSID and password.
If using 5 Ghz or 2.4 Ghz channel 12+, also double check the wifi country code is set correctly.

DdraigX, 2023-06-20:
When I started, it had pulled it from the registry settings in some type of hash value? So I manually typed everything over for both the Wifi and username/password. 

I copied an older wpa_supplicant.conf with the correct info over and that worked for that test, but I couldn't get in because it didn't create the pi user password properly. That's when I realized it wasn't just my typing an issue, also did show password and that was correct.

DdraigX, 2023-06-20:
We can mark this as resolved I suppose. I went through and hand typed all the information making sure case was correct, and having wiped out any saved pre-configuration and now it seemed to work.

closed 2023-06-20 by DdraigX