raspberrypi/rpi-imager #598: The scrolling is wonky

TojikCZ, 2023-06-20:
Hi, I have an XFCE setup and my scrolling is all wonky.
In the advanced settings menu, it takes two clicks of the scroll wheel sufficiently close together to smooth scroll across the whole dialog
In wifi country and locale settings, the letter keys don't work (search by first letter), scrolling by the mouse wheel is extremely slow, so holding an arrow key ends up being the only usable option.
I use the imager in bursts, but when I need to use it, these are the absolute worst parts of it for me.
Other than that, it's pretty great. Thank you for this free tool

maxnet, 2023-06-20:
>I have an XFCE setup

If your window environment is not on speaking terms with Qt, and does not passes the system settings to it, you may try if it works better if you install "qt5ct" and adjust "mouse wheel scroll lines" there manually.

KernelBypass, 2023-07-03:
Same issue on Windows 10.

TojikCZ, 2023-07-03:
Started to use the dropdowns as edit boxes. Unscrollable

tdewey-rpi, 2023-10-19:
I can confirm I've seen the same issue with the Qt controls on Windows 11. One workaround is to treat your mouse cursor like a virtual touch input - so click and drag the combobox, etc to scroll smoothly across multiple items.

Unfortunatately, however, I have not yet discerned what causes this, so I'm going to keep this one open pending further investigation.

tdewey-rpi, 2024-09-19:
I believe this is largely addressed in 1.9.0 - @TojikCZ are you able to run the AppImage and confirm?

TojikCZ, 2024-09-19:
I agree, now the page scrolls correctly. I still wish that the wifi country list would jump to the first country code that starts with a letter I press on the keyboard, or be otherwise more browseable, it is a chore to scroll through the whole thing on a non freewheeling mouse, but the issue I reported is resolved. Feel free to close. Thank you

tdewey-rpi, 2024-09-19:
You _should_ be able to treat the combobox as a text box, and start typing a country code (if you know it). There should even be autocomplete. I just tested the macOS release for that behaviour, and might be more usable.

Closing as fixed for now.

closed 2024-09-19 by tdewey-rpi