raspberrypi/rpi-imager #599: Off Topic

actraiser30, 2023-06-21:
Hello Maxnet i hope you dont mind posting here this

i was wondering if you can give me your berryboot github page in case that project is abandoned you dont have time for this etc.

I dont know for sure how to do commits and other stuff but i have made a partial berryboot update,i updated everything except the kernel its not possible since the build fails every time.

im very interesting about improving berryboot and the project in general has many potential 

thanks in advance

maxnet, 2023-06-21:
>i was wondering if you can give me your berryboot github page

No, sorry.

And Berryboot has nothing to do with Imager.
So closing this.

closed 2023-06-21 by maxnet

actraiser30, 2023-06-21:
> > i was wondering if you can give me your berryboot github page
> 
> No, sorry.
> 
> And Berryboot has nothing to do with Imager. So closing this.

i know it has nothing to do i just did not know if you will see it if i post in berryboot,at least can you tell me if you will update the berryboot code/instructions etc so the build wont fail anymore?

thanks