raspberrypi/rpi-imager #600: Unable to build on Apple M2

hasiflo, 2023-06-22:
Hi all,

I tried to build the rpi-imager on a Macbook Pro with M2 silicon. Has someone of you successfully built this project before on a M2?
If yes, have you modified something in the CmakeLists.txt or did it work out of the box?

I've downloaded and installed Xcode and QT Creator and have followed the readme for building the application. 
When using the suggested Qt 5.x version (Qt 5.15.2 in my case) and build the Debug Version for "Desktop (arm-darwin-generic-mach_o-64bit)", I get following error:

`Undefined symbols for architecture arm64:
  "qt_assert_x(char const*, char const*, char const*, int)", referenced from:
      QList<QString>::operator[](int) in lto.o
      QList<QString>::at(int) const in lto.o
      QList<DriveListItem*>::at(int) const in lto.o
      int qRegisterNormalizedMetaType<std::__1::vector<Drivelist::DeviceDescriptor, std::__1::allocator<Drivelist::DeviceDescriptor>>>(QByteArray const&, std::__1::vector<Drivelist::DeviceDescriptor, std::__1::allocator<Drivelist::DeviceDescriptor>>*, QtPrivate::MetaTypeDefinedHelper<std::__1::vector<Drivelist::DeviceDescriptor, std::__1::allocator<Drivelist::DeviceDescriptor>>, QMetaTypeId2<std::__1::vector<Drivelist::DeviceDescriptor, std::__1::allocator<Drivelist::DeviceDescriptor>>>::Defined && !QMetaTypeId2<std::__1::vector<Drivelist::DeviceDescriptor, std::__1::allocator<Drivelist::DeviceDescriptor>>>::IsBuiltIn>::DefinedType) in lto.o
      QVector<unsigned long>::at(int) const in lto.o
      QList<QByteArray>::operator[](int) in lto.o
      QList<unsigned int>::erase(QList<unsigned int>::iterator) in lto.o
      ...`

I also tried to build with Qt 6.3.2 for macOS, but there I get the following error:

`WARNING: QML file parse error (deployment will continue):
WARNING: "qmlimportscanner: No such file or directory: \"/Users/..../Downloads/rpi-imager-1.7.5\\ 2/src\"\n"
ERROR: qmlimportscanner output error. Expected json array, got:
ERROR: ""
ninja: build stopped: subcommand failed.`

Would be happy if someone has a hint for me ;)
Thanks!

maxnet, 2023-06-22:
Unless you have a commercial Qt license, you need to use Qt 6+ to build for M1/M2.

Not certain what is up with your qmlimportscanner, but "/Users/..../Downloads/rpi-imager-1.7.5\\ 2/src\" does show you have a slash and a space in your rpi-imager folder name, and you may want to try if it builds better if you put the files in a folder without such spaces and special character.

Also be aware that if you want to be able to run the program you build on any other Mac than your own, you also need a paid Apple developer subscription, and get the thing code signed/notarized at the end.

hasiflo, 2023-06-26:
Thanks @maxnet,

your hint was right. I changed the folder name without any spaces. Now it is building at least a package with ~400MB (built  for macOS with Qt 6.3.2).
But then I get an error at execution of the program. 

`/Users/.../Downloads/rpi-imager-1.7.5/build-src-Qt_6_3_2_for_macOS-Debug/rpi-imager.app/Contents/MacOS/rpi-imager ; exit;
user@MacBook-Pro ~ % /Users/.../Downloads/rpi-imager-1.7.5/build-src-Qt_6_3_2_for_macOS-Debug/rpi-imager.app/Contents/MacOS/rpi-imager ; exit;
QML debugging is enabled. Only use this in a safe environment.
Found invalid metadata in lib /Users/.../Downloads/rpi-imager-1.7.5/build-src-Qt_6_3_2_for_macOS-Debug/rpi-imager.app/Contents/PlugIns/platforms/libqcocoa.dylib: Invalid metadata version
qt.qpa.plugin: Could not find the Qt platform plugin "cocoa" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
` 

The libqcocoa is inside the package (PlugIns/platforms). I have installed Qt 6.3 using the Qt MainenanceTool. Do you have a hint for me on how to further debug this or how to get a correct version of the plugIn?

Cheers!

maxnet, 2023-06-26:
>Now it is building at least a package with ~400MB (built for macOS with Qt 6.3.2).

While M1/Intel universal builds do are bloated by design, 
400 MB is a bit much.
You may want to set the current build configuration to "release" instead of "debug" in Qt creator and rebuild.

hasiflo, 2023-06-26:
I did a clean rebuild with the "release" configuration and now I got a package of about 340MB. The problem still is the same with the "Invalid metadata version qt.qpa.plugin".
Any suggestions on this problem?

tdewey-rpi, 2024-09-18:
I can confirm I'm happily building on Apple Silicon - and I see .App directories of the order of ~400MiB and .dmgs of around ~130MiB.

Closing this as stale - if there's still issues building against macOS, please raise new issues against 1.9.0

closed 2024-09-18 by tdewey-rpi