raspberrypi/rpi-imager #601: Unsure of how to build on windows

oxmc, 2023-06-30:
I am having difficulties building the project on Windows due to a specific error message that appears during the linking phase, and i'm unsure of how to address it.

I am using a custom version of the rpi-imager code which is located at: [https://github.com/HooverHigh/hooverhigh-imager](https://github.com/HooverHigh/hooverhigh-imager)

The error message I receive is as follows:
```
[ 85%] Linking CXX executable hooverhigh-imager.exe
The system cannot find the path specified.
make[3]: [C:\Users\Fallo\AppData\Local\Temp\cc14E2Cf.mk:3: C:\Users\Fallo\AppData\Local\Temp\ccY6KXnX.ltrans0.ltrans.o] Error 1 (ignored)
The system cannot find the path specified.
make[3]: [C:\Users\Fallo\AppData\Local\Temp\cc14E2Cf.mk:12: C:\Users\Fallo\AppData\Local\Temp\ccY6KXnX.ltrans3.ltrans.o] Error 1 (ignored)
The system cannot find the path specified.
make[3]: [C:\Users\Fallo\AppData\Local\Temp\cc14E2Cf.mk:15: C:\Users\Fallo\AppData\Local\Temp\ccY6KXnX.ltrans4.ltrans.o] Error 1 (ignored)
```

OS info:
[image]

cmake version 3.27.0-rc3

Please note i have almost no idea how to compile on windows, my linux build does work and function, just having issues with windows.

oxmc, 2023-06-30:
> > I am having difficulties building the project on Windows...
> > I am using a custom version of the rpi-imager code...
> 
> Have you tried building the standard version of the rpi-imager code (i.e. from this repo) on Windows?

No, I have not, I will try that first.

oxmc, 2023-07-02:
Bumping

oxmc, 2023-08-06:
Unsure if I'm doing this incorrectly or what not

maxnet, 2023-08-06:
Has only been tested with 32-bit mingw that comes with Qt.
Not with any ither toolchain.

oxmc, 2023-09-02:
@maxnet Update,

I got most of the issues fixed, my only issue is now with OpenSSL:

```
C:\Users\Fallo\AppData\Local\Temp\cc1W7uXb.ltrans1.ltrans.o:<artificial>:(.text+0x7b8): undefined reference to `SHA256_Update'
C:\Users\Fallo\AppData\Local\Temp\cc1W7uXb.ltrans1.ltrans.o:<artificial>:(.text+0x72b7): undefined reference to `SHA256_Final'
C:\Users\Fallo\AppData\Local\Temp\cc1W7uXb.ltrans2.ltrans.o:<artificial>:(.text+0x9cd): undefined reference to `SHA256_Update'
C:\Users\Fallo\AppData\Local\Temp\cc1W7uXb.ltrans2.ltrans.o:<artificial>:(.text+0x926a): undefined reference to `SHA256_Update'
```

maxnet, 2023-09-02:
Make sure you build OpenSSL from source with mingw and are not using a vc++ version.

See also: https://github.com/raspberrypi/rpi-imager/issues/351#issuecomment-1031517167

oxmc, 2023-09-02:
@maxnet Update:

I have compiled openssl, and it looks ike most of my issues with that are gone, except linking 2 files:

```
Error copying file "D:/Files/qt/5.14.2/5.14.2/mingw73_32/lib/cmake/Qt5/../../../bin/libssl-1_1.dll" to "C:/Users/Fallo/Documents/Github/hooverhigh-imager/build/deploy".
Error copying file "D:/Files/qt/5.14.2/5.14.2/mingw73_32/lib/cmake/Qt5/../../../bin/libcrypto-1_1.dll" to "C:/Users/Fallo/Documents/Github/hooverhigh-imager/build/deploy".
mingw32-make.exe[2]: *** [CMakeFiles\hooverhigh-imager.dir\build.make:673: hooverhigh-imager.exe] Error 1
mingw32-make.exe[2]: *** Deleting file 'hooverhigh-imager.exe'
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:400: CMakeFiles/hooverhigh-imager.dir/all] Error 2
mingw32-make.exe: *** [Makefile:135: all] Error 2
```

maxnet, 2023-09-02:
Male sure you copied the openssl dlls you compiled to ../bin
(And the libs to

oxmc, 2023-11-24:
@maxnet I feel too dumb to fully understand it, but i installed qt, added the openssl stuff, but it just wont compile, something about deprecated ssl function names, i dont know what to look for or continue, are you able to provide me with build commands in the cli or a preview of compiling the imager?

tdewey-rpi, 2024-09-19:
Closing this primarily as stale, but also because I don't have the bandwidth to support forked versions of the project.

If this is still an issue, oxmc, I'd encourage you to sync to 1.9.0 and follow the current build instructions.

closed 2025-06-05 by tdewey-rpi