raspberrypi/rpi-imager #602: Error during verification for fedora server 38.

LOLwierd, 2023-07-02:
Constantly getting MBR invalid signature error during verification while writing Fedora server 38 to my SD card.

Using RPI Imager on Windows.

johnyc90, 2023-07-31:
I'm having exactly the same error creating an image using RPI-Imager v1.7.5 on Fedora 38. Not sure the error comes from rpi-imager or the fedora image.

maxnet, 2023-07-31:
Please be very specific about the image you are using.
You do are using their aarch64 *RAW* image, and not the ISO (which is NOT for the Pi, but for other ARM devices)?

johnyc90, 2023-07-31:
@maxnet you're absolutely right... I've used the iso. Downloading the raw image right now.

johnyc90, 2023-07-31:
Worked for me! thanks @maxnet

maxnet, 2023-08-01:
Good to hear, it works now.

I am using the topic starter (@LOLwierd) had the same problem, and am closing this.
If not, just re-open.

closed 2023-08-01 by maxnet