raspberrypi/rpi-imager #605: Question: Can I export/import the Advanced options?

samuelT2, 2023-07-12:
Hi there

The imager is a great piece of software which I use several times a month. And since you brought the Advanced options, it's even easier to setup WLAN/SSH etc.

Is there a way to export the Advanced options saved to a file? Perhaps from the registry or somewhere?

I'm using the latest verson on Windows 10.

Thank you very much for your support!

MrSmoer, 2023-08-21:
I guess so, yes.
The imager uses Qt's QSettings to store the settings and according to the [documentation](https://doc.qt.io/qt-5/qsettings.html#locations-where-application-settings-are-stored) the default for their example is "HKEY_CURRENT_USER\Software\MySoft\Star Runner", for the Imager I found the settings at
`HKEY_CURRENT_USER\Software\Raspberry Pi\Imager`
and especially the advanced settings you were looking for at
`HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization`

samuelT2, 2023-08-21:
Excellent, that worked. Thank you for your help!

closed 2023-08-21 by samuelT2