raspberrypi/rpi-imager #606: Can't write to /dev/mmcblk0 on Arch Linux: is a system drive

tarneaux, 2023-07-12:
I can't write an image to my SD card on `/dev/mmcblk0` because RPi-imager thinks it's a system drive.
Writing with the CLI versio with `--enable-writing-system-drives` works but the CLI doesn't allow me to add wifi, SSH and username/password settings.

Possible solutions:
- Make system drive detection less agressive. I don't know why this would be a system drive if it's at `/dev/mmcblk0` which doesn't look much like `/dev/sda` or `/dev/nvme0`...
- Add a shorcut on the system drive selection to show system drives (and add it to documentation :-))
- Allow using the `--enable-writing-system-drives` option with the GUI

maxnet, 2023-07-12:
Recall that on Linux we only display drives that are marked as removable or hotplug.

Does `lsblk -o name,rm,hotplug` show zeroes for the device?

tarneaux, 2023-07-13:
> Does `lsblk -o name,rm,hotplug` show zeroes for the device?

Indeed:
```
NAME        RM HOTPLUG
...
mmcblk0      0       0
├─mmcblk0p1  0       0
└─mmcblk0p2  0       0
```

lurch, 2023-07-13:
Just out of curiosity, which hardware and OS are you running on?

tarneaux, 2023-07-13:
My laptop (desktop has no SD card reader) is an Asus R558UQ that's a few years old.

GitTrialdude, 2023-07-27:
If I am trying this from the embedded Linux on a ChromeOS device, will I have the same issues?  The drive shows up in /mnt/ChromeOS/removable and I get the 0's as well.  Is it not showing up due to it mounting to /mnt?

maxnet, 2023-07-27:
>If I am trying this from the embedded Linux on a ChromeOS device, will I have the same issues? 

No, different issue.
Being that Linux apps run in a virtualized environment under ChromiumOS, and the raw disk device is not currently made available under that.

https://issuetracker.google.com/issues/260278133?pli=1

lurch, 2023-08-16:
Possibly related to the discussion in #610 about ArchLinux?

ToxicFrog, 2023-08-17:
I'm not running Arch but I suspect it's the same underlying problem, yes.

tarneaux, 2023-09-15:
Here's a quick and dirty fix for anyone having the same issue: https://wiki.archlinux.org/title/udev#udevd_hangs_at_boot

fallingcats, 2023-10-02:
> Here's a quick and dirty fix for anyone having the same issue: https://wiki.archlinux.org/title/udev#Some_devices,_that_should_be_treated_as_removable,_are_not

I've tried that workaround and it does not seem to help:

```
$ udevadm info /dev/mmcblk0 | grep UDISKS
E: UDISKS_AUTO=1
E: UDISKS_SYSTEM=0
$ lsblk -o name,rm,hotplug               
NAME        RM HOTPLUG
mmcblk0      0       0
nvme0n1      0       0
├─nvme0n1p1  0       0
├─nvme0n1p2  0       0
└─nvme0n1p3  0       0
```

Any ideas?

umlaeute, 2023-10-02:
for me (Debian/unstable amd64) , rpi-imager-1.7.6 fixes the issue.

the problem with that is of course, that there is no rpi-imager-1.7.6 release yet, but cloning the repository (e8931b7b8f7dfb72ce0547258183330ea998acdc) and building locally gives me an rpi-imager that can see my bulit-in SDcard writer.

umlaeute, 2023-10-02:
i still think that having an option `--enable-writing-system-drives` in GUI-mode would make sense.
it's obviously a dangerous option (so you have to explicitly enable it), but this issue is somewhat orthogonal to #610 (it's just that we could use `--enable-writing-system-drives` to workaround the other issue)

sudo-splinter-cell, 2023-10-18:
I couldn't build the new version under Raspberry Pi. I am getting error messages. Can anybody drop me the "deb" file for the latest version?

umlaeute, 2023-10-19:
How about mentioning the ominous error messages instead (probably in a new ticket)?

tdewey-rpi, 2023-10-19:
This should be resolved in v1.8.1, which is available under the published releases.

closed 2023-10-19 by tdewey-rpi

sblantipodi, 2023-10-23:
I still have this issue with 1.8.1 on both Windows 11 and Linux Ubuntu 23.10

emorgado, 2024-02-06:
The problem persist, I'm using version 1.8.5 on ubuntu  Ubuntu 22.04.3 LTS.
Linux ... 6.5.0-15-generic #15~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 12 18:54:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux