raspberrypi/rpi-imager #608: Advanced settings fail silently if no password set

chrullrich, 2023-07-16:
The advanced settings dialog does not give any useful feedback when it does not save settings on account of the user password field being blank. The only thing I could find is that the field label turns red. There is no error message indicating to the user that their input has actually not been saved. (The save button also sets the input focus to the empty password field, but because the window a) does not bother to scroll that field into view, and b) is needlessly tiny and not resizeable, this nice gesture is frequently lost on the user.)

This is bad UI because the user cannot predict the behavior of the window; it is not clear that the window is supposed to close after clicking "Save" and that it is therefore a sign of trouble when it does not. For some users, manually closing the window after apparently saving the settings may make sense, and there is no prompt to save modified settings when closing the window either.

If the UI was designed more in line with platform standards and put less emphasis on being white, red, and flat, that might be more obvious. Changing the caption of the "Save" button to something that typically indicates the save-and-close operation on the respective platform, such as "OK" on Windows, would also help. The user is going to interact with the UI for a very few minutes before starting a possibly lengthy operation (writing the image). It would be nice of the application to make the utmost effort to ensure the user gets the result they wanted.

hagenbuch, 2023-10-10:
Wow.. this remained untouched since 2016!

maxnet, 2023-10-10:
Problem with error messages is that they need to be translated in 15 languages.
Red = trouble is pretty universal.

Scrolling issue should no longer be there with tabbed settings window.

lurch, 2023-10-12:
> Wow.. this remained untouched since 2016!

If you mouse-hover over the date (where it says "Jul 16") the tooltip says "Jul 16, 2023, 10:58 AM GMT+1", which is _much_ more recent than 2016 :wink:

hagenbuch, 2023-10-12:
Well I agree! Sorry. Not all countries have the day after the month :)

tdewey-rpi, 2024-09-18:
[image]

1.9.0 will refuse to let you save, and highlight the field that failed.

closed 2024-09-18 by tdewey-rpi