raspberrypi/rpi-imager #609: Linux repository . http instead of https after flash sd card or usb key

sympaticBE, 2023-07-16:
Hello,
It's my first post on  githhub. Please be nice.

I've got a strange problem with pi-imager 

### Situation:

- OS : Debian linux distribution or on w10
- Version: Pi-imager v1.7.5
- Storage media : sd card or usb key
- For Flash : Raspberry Pi OS (any version)

### Problème
After configuring and installing raspery pi os ( User mod de passe,...)
The updates do not work because the repository are in http and not https and the security certificates are not validated.

Any idea where the problem is coming from?

Ps: I am a beginner in free software

maxnet, 2023-07-16:
```
The updates do not work
```

How are you trying to apply updates exactly?
If you open a terminal window and type ```sudo apt update && sudo apt dist-upgrade```, what exact error message do you get there?

```
because the repository are in http and not https
```

Not sure where that conclusion comes from.
Debian's repository files have their own cryptographic signing mechanism, ensuring the integrity of the files.
No need for https for that.

(Only downside of the repository being http, do is that entities able to intercept your network communications do can see which packages you are installing.)

sympaticBE, 2023-07-16:
Hi,

#How are you trying to apply updates exactly?
Directly at the end of the installation process raspbery pi os.
Also
Via 
sudo apt update && upgrade
Via
App recomended software

> The Pi does show the correct date and time when you enter date?

Yes, I had checked that.

> Not sure where the conclusion comes from that it fails because the repository is http...

I thought that when I saw the error messages mentioning signatures

Here's my error message :

> Réception de :1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
> Err :1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
>   Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> Réception de :2 http://archive.raspberrypi.org/debian bullseye InRelease
> Err :2 http://archive.raspberrypi.org/debian bullseye InRelease
>   Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> Lecture des listes de paquets... Fait
> E: Impossible de récupérer http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> E: Le dépôt http://raspbian.raspberrypi.org/raspbian bullseye InRelease n'est plus signé
> N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
> N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.
> E: Impossible de récupérer http://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> E: Le dépôt http://archive.raspberrypi.org/debian bullseye InRelease n'est plus signé
> N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
> N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.

maxnet, 2023-07-16:
>I thought that when I saw the error messages mentioning signatures

It is expecting to download a file from the repository that looks like:

```
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Origin: Raspbian
Label: Raspbian
Suite: stable
Codename: bullseye
Date: Sun, 16 Jul 2023 16:21:56 UTC
Architectures: armhf
Components: main contrib non-free rpi firmware
Description: Debian armhf testing distribution for Raspberry Pi
MD5Sum:
 c9b93d0f30bc31f286fea8c40e2f2afa 68694624 main/binary-armhf/Packages
 638510a1eb3b1c3e029f180f032579d8 18257320 main/binary-armhf/Packages.gz
 37a3b9c411c12bf678b0248b6a7920f9 13232868 main/binary-armhf/Packages.xz
 3ae69d1cbef047fef5641695e84a96d0 149 main/binary-armhf/Release
 6538f60494a6ccdbff098ea8ef726e0d 260870 main/debian-installer/binary-armhf/Packages
 57d0defc0947372ef8bd43496aac3b30 76938 main/debian-installer/binary-armhf/Packages.gz
 70bc3b834aba29f996a31f49593799db 63624 main/debian-installer/binary-armhf/Packages.xz
 425078d08e70cf7fe9e27be659e98a82 57799196 main/source/Sources
 8b79e8d73c7f5d128f5b37ec7412df36 15418721 main/source/Sources.gz
 5c56c5984357467a956fca2f4eadef83 12225488 main/source/Sources.xz
 4cec8dca27312eb805542c95c4fdb930 150 main/source/Release
 0e43fb8310b603b866dcf8101cbca723 242223 contrib/binary-armhf/Packages
 b9f63dfe02b132f97f464df21961a8a3 70253 contrib/binary-armhf/Packages.gz
 8560a22813efa31c86d790fcdc5bfe22 60184 contrib/binary-armhf/Packages.xz
 ef76b4bdd97d38604ef128b5d3830f59 152 contrib/binary-armhf/Release
 5498597e6f5fd7a5ac2e6436a48bbea7 332663 contrib/source/Sources
 a26fc680fb0efd2a26f54cd04eef326b 96891 contrib/source/Sources.gz
 a57892cb85b97f1ad26bcd7aad5221ee 82168 contrib/source/Sources.xz
 3dffb55075d66dee65ccd6e3569f1595 153 contrib/source/Release
 725c9c321fb0a06b39559c73f2c6a048 531404 non-free/binary-armhf/Packages
 f8af2c9fd6f3b0ae3d78fe3591b297e0 129847 non-free/binary-armhf/Packages.gz
 2ed250c0e5973135cd6748d0a3bce685 106300 non-free/binary-armhf/Packages.xz
 6699b0874d711b697b7bc047f1b05f1f 153 non-free/binary-armhf/Release
 26a755833e2760b361b2b9fd897967f2 611229 non-free/source/Sources
 92fc55cfe9f406cf6a9f2fec4292f18f 167683 non-free/source/Sources.gz
 113160d75ad99d3d48d3609ac4e42bc3 140712 non-free/source/Sources.xz
 baff18ca7792096310768d0f46b6c3db 154 non-free/source/Release
 234ca30c67ce47e0f0d2c0acd2ff845d 2704 rpi/binary-armhf/Packages
 607291567cdbf85df558b42440a3c781 1299 rpi/binary-armhf/Packages.gz
 fcb836584d1ddd17cce4d8e8b9204b57 1360 rpi/binary-armhf/Packages.xz
 7aba182e24e4c7c69c23ab20b2df9f9d 148 rpi/binary-armhf/Release
 009bfc1eb86de04b6a2fe3a30f9b5d9d 2437 rpi/source/Sources
 9a4c577a8cf10f9f5b6ea2cfdf6b2d43 1097 rpi/source/Sources.gz
 96df10ea1adbae8dadc07c5288e7a7fc 1132 rpi/source/Sources.xz
 ebd854cd0ef22860b2f633d17c0ddc4a 149 rpi/source/Release
 463da616978c386d6802c96ca0fbf240 4243 firmware/binary-armhf/Packages
 520fbaf9233c456008ad30bfb8de5072 1201 firmware/binary-armhf/Packages.gz
 6eb9976d64bd3386e19b2b399a8581e3 1232 firmware/binary-armhf/Packages.xz
 d062cc51ac9addc8dc3471814b7084cc 153 firmware/binary-armhf/Release
 40f2153b1a1bd229e69305fa5be39a7b 1681 firmware/source/Sources
 7b75cec7eb703329236f593be9cd9c9e 677 firmware/source/Sources.gz
 ec5343a68232fd0900a355f0c3b8392f 736 firmware/source/Sources.xz
 8ea82219c23d23776329a49731e6c940 154 firmware/source/Release
 ad8051ae61d7b4605f351149ec148bf4 631303489 main/Contents-armhf
 a04e15f0fb56406bba1e7d36f5fcb11c 41278508 main/Contents-armhf.gz
 95e9d407c1a9e918ffdd3e8d5f59494c 2738145 contrib/Contents-armhf
 00d66f14d2a1c85807c698bf5888cf20 217860 contrib/Contents-armhf.gz
 68fac21ab410a1af3ac69a846cd48163 17464321 non-free/Contents-armhf
 5c9cf6ae42a35ff16d6fabd1a150323d 907865 non-free/Contents-armhf.gz
 8fd85a28d7ccaa1fa9594ff4eddfd801 1099 rpi/Contents-armhf
 3ad2b55778385f9cf45f1a9b0243362e 307 rpi/Contents-armhf.gz
 0f02c7bbd26ef385b307d6d352935ca0 51588 firmware/Contents-armhf
 3ee267035ba7d5e452c3023043c55748 4152 firmware/Contents-armhf.gz
[... 122 lines cut ...]
 b217e83897f860509b732de849639d13797caba26f6258b0813ecc5c6814322a 332710 main/Contents-udeb-armhf
 3809d3dc6957ac650a7da2a998a8a4695cf3bd9ce55f5eb1915c5819c0c2b320 30005 main/Contents-udeb-armhf.gz
 b217e83897f860509b732de849639d13797caba26f6258b0813ecc5c6814322a 332710 uContents-armhf
 3809d3dc6957ac650a7da2a998a8a4695cf3bd9ce55f5eb1915c5819c0c2b320 30005 uContents-armhf.gz
-----BEGIN PGP SIGNATURE-----

iQEzBAEBCAAdFiEEoNo40Ndui11jiHKBkWWTjZD93S4FAmS0GSQACgkQkWWTjZD9
3S6NrQgAhBDl7uYl6tx0wimfrbya1qn3DDsKMmQIgf7wh4Boa5fDwqBHblxXC79i
O9l1aAPMAYRFXJs2h5TlPPxOUJDWQerbnG/CKqChJS/9NQoIEjPg7O4q2+ZpqKpX
rWI8oPsPInkAcmcyfQipLNVVS9u8WwYxZ8YLkVOGXt0qKv/1n5nEd6cpD1menIbc
cL5EeJ8iFKPMxb39xwCT9zzhZIMIWfg6jLO9fV2vP1oyzSxjPbnXD9ovVDFXX6Mn
dWwilfibsyLh9GEnb9K1FRfIS3w/9QrICqvbeuXyOMc5NT3jlrIFUmnKreQK24x7
PwJy0FCBDtnrl5jIlkULMaqZyc6szw==
=/xdr
-----END PGP SIGNATURE-----
```

So it is a file that is split into two parts.
The actual data, and the PGP signature.
The error "NoSplit" means what it received over the network did not start with "-----BEGIN PGP SIGNED MESSAGE-----", but it received something else...

You have a normal home network, in which the Pi does have Internet access directly, without having to login to some wifi portal first?
It is not a corporate network, and there is not any content filtering/blocking being done at network level?

What HTTP responses do you get if you do: `LANG=en sudo apt update -o Debug::Acquire::http=true` ?

sympaticBE, 2023-07-17:
Good evening,

Thank you very much for your questions and suggestions. Allowed me to solve the problem.

**How :**

Your previous question:

> You have a normal home network, where the Pi has direct access to the Internet, without having to connect to a wifi portal first?
> It's not a corporate network, and there's no content filtering/blocking at network level?

No, it's a home network.
Modem router --> Switch 5 port --> PI4

And your question

> What HTTP responses do you get if you do: LANG=en sudo apt update -o Debug::Acquire::http=true ?

They made me look at my IP address and ask about my local network.
I remembered that I had used this Pi4 for a self-hosting test. I had blocked the Pi's mac address to give it an IP Fix and I had given it a machine name in the modem.

I think that as the PiOS gives the name raspberyPI by default that generated a conflict because the name in the modem was different.

I removed the mac filtering and the name in the router. 
And bingo!
The update works.

All this motivates me to do this training in network and cybersecurity basics.

Because I'd like to understand the network layers better and understand this kind of bug from the point of view of the osi model.

Thanks again

**For your information, here is the result of the command** 

LANG=en sudo apt update -o Debug::Acquire::http=true

> LANG=en sudo apt update -o Debug::Acquire::http=true
> 0% [En cours]GET /raspbian/dists/bullseye/InRelease HTTP/1.1
> Host: raspbian.raspberrypi.org
> Cache-Control: max-age=0
> Accept: text/*
> If-Modified-Since: Tue, 02 May 2023 22:24:33 GMT
> User-Agent: Debian APT-HTTP/1.3 (2.2.4)
> 
> 
> GET /debian/dists/bullseye/InRelease HTTP/1.1
> Host: archive.raspberrypi.org
> Cache-Control: max-age=0
> Accept: text/*
> If-Modified-Since: Tue, 02 May 2023 10:12:50 GMT
> User-Agent: Debian APT-HTTP/1.3 (2.2.4)
> 
> 
> Answer for: http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease
> HTTP/1.1 200 OK
> Server: nginx
> Date: Mon, 17 Jul 2023 16:18:02 GMT
> Content-Type: text/html
> Transfer-Encoding: chunked
> Connection: keep-alive
> X-Frame-Options: SAMEORIGIN
> Cache-Control: public, max-age=31536000
> 
> Réception de :1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
> Err :1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
>   Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> 0% [Attente des fichiers d'en-tête]Answer for: http://archive.raspberrypi.org/debian/dists/bullseye/InRelease
> HTTP/1.1 200 OK
> Server: nginx
> Date: Mon, 17 Jul 2023 16:18:03 GMT
> Content-Type: text/html
> Transfer-Encoding: chunked
> Connection: keep-alive
> X-Frame-Options: SAMEORIGIN
> Cache-Control: public, max-age=31536000
> 
> Réception de :2 http://archive.raspberrypi.org/debian bullseye InRelease
> Err :2 http://archive.raspberrypi.org/debian bullseye InRelease
>   Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> Lecture des listes de paquets... Fait
> E: Impossible de récupérer http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> E: Le dépôt http://raspbian.raspberrypi.org/raspbian bullseye InRelease n'est plus signé
> N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
> N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.
> E: Impossible de récupérer http://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Le fichier signé en clair n'est pas valable, ce qui a été reçu est « NOSPLIT ». Peut-être le réseau nécessite-t-il une authentification.
> E: Le dépôt http://archive.raspberrypi.org/debian bullseye InRelease n'est plus signé
> N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
> N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.

closed 2023-07-17 by sympaticBE