raspberrypi/rpi-imager #611: Add .bin files to the listing

Awire9966, 2023-07-18:
Some OSes (Like ChromeOS Flex) come on .bin files that work with the imager, but arent part of the default file browser.

tdewey-rpi, 2023-10-19:
@Awire9966 If you're aware of a contact point for the OS you'd like to be added to the OS selector, please ask them to get in touch / raise an issue here. They'd need to provide a JSON listing, or negotiate to host their own sublist JSON.

Unfortunately, however, I don't have the bandwidth to chase down new images, so I'm going to mark this as closed.

closed 2023-10-19 by tdewey-rpi

tdewey-rpi, 2023-10-21:
Re-reading, @Awire9966, is your request specifically to be able to select .bin files in the custom flashing option?

Awire9966, 2024-03-06:
> Re-reading, @Awire9966, is your request specifically to be able to select .bin files in the custom flashing option?

My hope is to select bin files from the file explorer window that pops up. Add the bin files to the regex expression.

lurch, 2024-03-06:
Is a `.bin` file always a raw uncompressed disk-image file, the same as a `.img` file?