raspberrypi/rpi-imager #612: advance settings fail for ssh

manasmgkar, 2023-07-21:
flashing the image enabling ssh in the advance settings,while the pi connects to the wifi using the advance settings the ssh password keeps failing,i tried reflashing multiple times with different username pass combinations but ssh seems to fail always stating wrong password. Doing it the manual way works. 

Same error occurred for me on both pi4 and pi3 b devices.

kenil6497, 2023-08-17:
> flashing the image enabling ssh in the advance settings,while the pi connects to the wifi using the advance settings the ssh password keeps failing,i tried reflashing multiple times with different [...]

same issue, when you say doing it manually do you mean by flashing using Rufus?

sirhopcount, 2023-08-25:
I can also confirm this error, using version v1.7.5 of the imager.

tdewey-rpi, 2023-10-19:
@sirhopcount, @kenil6497, @manasmgkar,

Thanks for the report. Can you confirm which OSes you were attempting to write at the point you encountered this issue?

sirhopcount, 2023-10-20:
I believe it was the latest release (at that time) of the bullseye lite image.  I'll see if I have some time this weekend to verify.

tdewey-rpi, 2023-10-20:
Thanks for the response, @sirhopcount. If you get the opportunity, I'd appreciate you giving it a go with v1.8.1.

wombat2k, 2023-10-20:
Ran into the same issue. 

## Steps to reproduce
1. Used RPI Imager 1.8.1 on a Mac running Sonoma
2. Pick Ubuntu Desktop 23.10 (for rpi4)
3. Press shift-ctrl-x to get to Advanced settings. Enable ssh key with a public key
4. Flash
5. Boot
6. From your client machine, try to connect, you will get a connection refused on port 22

> ssh: connect to host rpi4 port 22: Connection refused

sirhopcount, 2023-10-22:
I'm a bit baffled as I'm unable to reproduce my initial issue. What I tested:

* RPI Imager version: 1.7.5

Settings:

* Enable SSH.
* Use password authentication.
* Set username and password.

## Hardware: Raspberry Pi 3 Model B Plus Rev 1.3

* Raspbian GNU/Linux 11 (bullseye) - Worked
* Raspbian GNU/Linux 12 (bookworm) -  Worked
* Ubuntu server 23.10 - Worked

## Raspberry Pi 4 Model B Rev 1.5
* Debian GNU/Linux 12 (bookworm) - Worked
* Debian GNU/Linux 11 (bullseye) - Worked
* Ubuntu server 23.10 - Worked
(All image were 64 bit)

The only difference between the tests and my initial attempt is the SD card size, the initial attempt being on a 32GB card and the tests being done with a 8GB card. Unfortunately I can't test with the 32G card as it is in use.

jeremyjjbrown, 2023-11-28:
I see this also on rasbian and a raspberrypi zero with some sandisk 64gb. It's literally my first experience with raspberrypi is broken horribly and I have to say my confidence levels are now very low since it fails to turn on ssh on a fresh install and I get a connection refused on port 22.

If there is something I can do to provide info let me know.

brianbolt, 2024-01-08:
> > Pick Ubuntu Desktop 23.10
> 
> Only Ubuntu Server supports customization.

Where does it say that?

tdewey-rpi, 2024-01-09:
Believed resolved in 1.8.4, where an issue around password inputs would have resulted in the first character being trimmed on the edit of an existing password. The subsequent reports of success and failure tie to this issue.

For adjacent issues, please raise new bugs.

closed 2024-01-09 by tdewey-rpi