raspberrypi/rpi-imager #613: Use user's account for public-key authentication, not the "admin" account

bit-bash, 2023-07-21:
On our systems, we run as normal users.
So, when I start the Imager, it requires admin authentication which is a different user account.
To enable SSH public-key authentication, it seems that the Imager is trying to find the admin account's id_rsa.pub instead of the user's own id_rsa.pub file - so typically the field is empty my admin account doesn't have a .ssh directory.
It seems more useful to either file the user's .ssh directory or allow the Imager to ask which user's .ssh directory to look for.
Of course, I can open the user's id_rsa.pub file in notepad and copy/paste it into the Imager, but this seems cumbersome.
Thanks!

maxnet, 2023-07-21:
The code to copy SSH public keys was written for Linux and OS X users, on which Imager runs as normal user.
May indeed not work correctly on Windows.

tdewey-rpi, 2024-09-18:
Closing as stale - the OP found a workaround, and I'm not convinced that allowing both password-authentication and public-key-authentication on a written OS is a sensible move.

closed 2024-09-18 by tdewey-rpi