raspberrypi/rpi-imager #614: Support `US - International` keyboard layout

CC007, 2023-07-26:
In the Netherlands we don't use the NL keyboard layout. We usually use the US - International layout (e.g. the US layout with dead keys). 

With xkb this would require `XKBLAYOUT` to be set to `us`, which is doable, but it also requires `XKBVARIANT` to be set to `intl`. This can unfortunately not be set right now in this rpi-imager:

https://github.com/raspberrypi/rpi-imager/blob/b49408781a3c347bd6f6c057c68bb34d6c06ad10/src/OptionsPopup.qml#L759

As you can see, the rpi-imager right now always sets it to be an empty string.

I would like for there to be an extra option to select the variant of the keyboard layout.

Here is the section of the UI that would need to change:
https://github.com/raspberrypi/rpi-imager/blob/b49408781a3c347bd6f6c057c68bb34d6c06ad10/src/OptionsPopup.qml#L337-L363

Also, for custom images, this would need to change to support the keyboard variant: 
https://github.com/raspberrypi/rpi-imager/blob/b49408781a3c347bd6f6c057c68bb34d6c06ad10/src/OptionsPopup.qml#L763

maxnet, 2023-07-26:
Not sure if there's much demand for this.

>In the Netherlands we don't use the NL keyboard layout. 
>We usually use the US - International layout (e.g. the US layout with dead keys).

Maybe that is still the case in some office environments.
But know few technical folks that use it, as it makes typing code with ' and " more work, and you need those more often than accented characters.
(with US international one needs to type ' + space to get a normal ' )

CC007, 2023-07-27:
Yes, that's the whole point. It allows for typing letters like ë, à and é, which are used in Dutch for many words. No one wants to remember or look up alt codes.