raspberrypi/rpi-imager #615: Ubuntu Server 23.04, with ssh configured, not working

wkhatch, 2023-07-27:
Using advanced settings to configure ssh when installing ubuntu server 23.04 64bit, setting username and password and toggling ssh to enable is not working; the resulting running pi does not accept ssh connections. Wifi configuration works fine, and am able to find the pi on the network using nmap, but subsequent attempts to connect via ssh time out.

wkhatch, 2023-07-27:
I should add this is on macox 13.2.1, using v1.7.5 of imager, and setting up username/password for ssh, not key based auth.

maxnet, 2023-07-27:
>but subsequent attempts to connect via ssh time out.

And that is the exact error message it gives?
Timeouts are more typically a symptom of connection problems (or firewall rules dropping ICMP packets).

If it failed to start the SSH daemon, you are more likely to get "ssh: connect to host x.x.x.x port 22: Connection refused" instead.

wkhatch, 2023-07-27:
I guess it just takes a long time for the sshd to spin up. I'd tried several times to connect, and got timeouts, and then, on a whim, just tried it again before attempting to reflash the card, and suddenly it worked.

closed 2023-07-27 by wkhatch