raspberrypi/rpi-imager #617: Hide Storage Volumes Option

Wade821, 2023-08-03:
Just an option to hide volumes as desired, as I have 0 interest in ever using the "Microsoft Storage Space Device" drives as media for the Raspberry Pi Imager. 

With perhaps a button to "Show Hidden Volumes" there that were manually hidden in order to unhide them.

[image]

tdewey-rpi, 2023-10-19:
Thanks for the report, @Wade821.

I think I can agree that writing to a grouped disk set like that isn't likely to be what someone wants to do, so filtering these out entirely seems like a solid first step. We can evaluate adding an option to show these in the UI at a later point.

Would that address your concern?

Wade821, 2023-10-20:
Yes, I think that would work just fine. Filtering them out to avoid accidental selection would be ideal.

tdewey-rpi, 2023-10-20:
Awesome, thanks for the response.

I'm waiting on a pair of drives to reproduce a Storage Space on my laptop, after which I'll be able to test locally. My initial thinking is that the `drivelist` module must be capable of filtering these in a straightforward manner, but without a reasonable test system I'm not prepared to start work.

tdewey-rpi, 2023-10-20:
If you've got `npm` ready to go, I'd love to see the output from `devicelist` on your machine - in particular, I want to know if it's flagging that the Storage Spaces are actually virtual disks or not. https://www.npmjs.com/package/drivelist

Wade821, 2023-10-20:
Ah, shoot. I'm on Windows primarily and it's been a long time since I played with WSL, as I'm using VMWare and I think that requires disabling the Hypervisor/'Virtual Machine Platform' stuff (and in turn the necessary components to play with WSL, since I recall having to turn that off to get VMWare to work).

I'm sure I could get that to work if really needed, but I'll have to do it later since that would take some time to reconfigure everything.

tdewey-rpi, 2023-10-23:
@Wade821, oof. Yes, don't let my request mess up your system. I'll get to this issue once I've got the drives.

tdewey-rpi, 2023-10-25:
My ordered drives arrived, but appear to have fallen foul of a poorly-documented requirement of Windows Storage Spaces - notably, that it requires drives that present 4K physical sectors. Unfortunately, mine present 512byte sectors, and no amount of gentle persuasion can convince it otherwise.

I've also tried creating virtual disks that would meet the criteria - but this _also_ failed because my laptop's internal drive _also_ uses 512byte sectors.

I'll try and source some other drives with 4K sectors to try this out.

Wade821, 2024-01-12:
Maybe if you don't find something soon, I'll get around to messing with my system and report back. I've been working on my own project with Github/Drawpile server stuff and need the existing virtual machines until that's completed. Hopefully not much longer.

twerthi, 2024-02-14:
Yes, please remove Storage Spaces drives from the selection list.  I don't want to get click happy when imaging my SD cards.