raspberrypi/rpi-imager #619: Outdated Ubuntu verson list

LoKy-dev, 2023-08-11:
I was trying to flash my sd card with Ubuntu Server 22.04.2 LTS (64-bit) but i kept getting 404 error.

Digging into the problem i found out that many user (#405) had the same issue.
The problem was that the imager was trying to fetch an old version and this might be the case too (the imager shows 22.04.2 but the official website show the 22.04.3).

Can this be fixed and is there a way to never run again in this issue?
I would like to implement myself but i don't know where to start

Imager version: 1.7.5

maxnet, 2023-08-11:
@waveform80 
https://changelogs.ubuntu.com/raspi/os_list_imagingutility_ubuntu.json lists http://cdimage.ubuntu.com/releases/jammy/release/ubuntu-22.04.2-preinstalled-desktop-arm64+raspi.img.xz which gives 404.

waveform80, 2023-08-11:
Argh ... let me ping the release team

waveform80, 2023-08-11:
Yup, someone forgot to (ping me to) run the update script for this bit. Should be fixed shortly.

closed 2023-08-13 by LoKy-dev