raspberrypi/rpi-imager #624: SD Card Overwrite Confirmation Dialogue Responses Incorrectly Inverted

astradamus, 2023-08-22:
[image]

I really can't imagine why this would be inverted. In English software, "Yes" goes on the left.

What possible good reason could there be for this? I've already misclicked it three times 😆

astradamus, 2023-08-22:
Incidentally, these buttons should really be more differentiated either war. Different colors, or no as gray and yes as red, something like this would be much safer.

maxnet, 2023-08-23:
>In English software, "Yes" goes on the left.

Depends whose UI guidelines you follow...

Apple MacOS UI guidelines: “A button that initiates an action is furthest to the right. The Cancel button is to the left of this button.

Google: "The dismissive action of a dialog is always on the left. Dismissive actions return to the user to the previous state.The affirmative actions are on the right. Affirmative actions continue progress toward the user goal that triggered the dialog.”

ghollingworth, 2023-08-23:
Yes, as Simon Long our esteemed UI engineer will tell you, in almost all languages we read from left to right, top to bottom (there are a few exceptions).  So the bottom right is the 'right place' for a completion button (kind of like a 'next page' would naturally be in a book)

Windows of course was first developed at a time when UI development was very young and they tended to stick with what they knew.  Other operating systems, I think, have switched to the bottom-right as it has become more obvious that that is the 'right-choice'!

Of course, most UI engineers agree that this should really be consistent with the platform, and therefore OK should be on the left on Windows but not on other OSes!

But I think consistency (within the application), is a better choice here...

astradamus, 2023-08-23:
I could see getting used to either one. It's the inconsistency that tripped me up, since as you said it's on Windows and that's what we're used to over there. I walked face-first into it 3 times in a row 😅

The highlighting differentiation would go a long way towards helping people from any configuration more quickly select the right option.

closed 2025-12-12 by tdewey-rpi