raspberrypi/rpi-imager #626: Imager fails, every time

w9ran, 2023-08-26:
I realize this is not a new issue but it is unresolved for many folks evidently.  I have never got it to work, not even once - despite trying three different card readers and a half dozen different TF cards (including premium ones).   Always the same result - verification error.   I finally gave up and used Etcher which works every time.   In the four hours I spent trying, I read many posts on various forums that claim it works perfectly and make condescending remarks to those having problems.  This doesn't seem to the compatible with the official attitude the Raspberry Pi Foundation takes toward users, and is especially undesirable when directed toward new users who encounter this Imager issue in their first attempt to use the Pi.
All I can say is - it's broken for some users, at least, and it would be greatly appreciated if some attention could be directed toward a permanent foolproof solution.  Thanks.

maxnet, 2023-08-26:
What OS were you running Imager on?
And since you have a working Pi now, does Imager work any better with the readers and SD cards you have, if you run it under RPI OS?

w9ran, 2023-08-27:
It was being run on Win 10.    Good thought to try it in a Pi -  I did so using one of the USB readers and cards that consistently failed in the PC, and it went to completion including verification without errors.      Maybe someone can determine what that might suggest.

p6002, 2023-08-27:
The app has been bugged for a long time. I am also currently creating an SD card for the fourth time hoping that this time it will work. 
It never works the first time that's for sure.

maxnet, 2023-08-27:
> It was being run on Win 10. Good thought to try it in a Pi - I did so using one of the USB readers and cards that consistently failed in the PC, and it went to completion including verification without errors. Maybe someone can determine what that might suggest.

That there is indeed nothing wrong with your readers and cards, but something is out-of-the-ordinary with your PC.
Not having your PC on my desk, I am afraid such things are very hard to debug.

There could be other (security) software writing to the SD card between the moment we finished writing to it, and read back all the contents to see if it is exactly the same as it is supposed to. Although that shouldn't be the case, as we zero out the partition table of the card, so it looks like there is nothing of interest there to other software.

Other than that there could also be hardware abnormalities. E.g. recall there were a few reports of people that it did work if they plugged the reader into a USB2 port on there computer, but not in a USB3 port. Or that a USB HUB in between caused problems.

TheMariday, 2023-09-18:
@w9ran, I faced a similar issue, I know this sounds insane but have you tried USB ports on the back of your machine?

https://github.com/raspberrypi/rpi-imager/issues/41

hagenbuch, 2023-10-13:
You mean for powering the Raspi? Actually I am using an extra power supply AND the power supply had been able to "carry" other Raspis, too.

I am still having the problem with a Raspi B+ I did run Rasbian desktop on it - and I used a SDcard capable of over 20 Mbyte/s (measures) and I tested it with f3write / f3read. I'm going to describe this in a new post.