raspberrypi/rpi-imager #627: Imager bug related to Orca (with detailed steps to reproduce)

Tinkerer555, 2023-08-26:
I was building images with desired packages installed to mimic the Pinn image raspios_64_full. Each time I thought I was done, I'd open Pi Imager to install the image on another card, and the Pi Imager would force close within a few seconds of selecting the computer directory to get to the image. 

I've spent the past few days teasing out which combination of updates and installs are breaking Imager and have narrowed it down to installing Orca from the Recommended Software collection. I will continue testing other base images to determine scope of impact, but I have one clear reproducible path to breaking Imager.

1 - install Raspberry Pi 64 with desktop from imager
2 - Update all packages (sudo[apt update][apt full-upgrade][apt clean][reboot])
3 - install Orca from app menu > Preferences > Recommended Software 
4 - reboot
5 - imager will now be broken and unable to navigate to a custom image

I have a an image with only Orca installed and Imager broken, and an image with all recomended packages installed except Orca with Imager still running fine. Both are Raspberry Pi Desktop builds with all packages updated.

My next steps will be to test on 64 without updating packages, and to test on 32 bit versions to see if that is a shared issue. 

My test Pi's have been Pi 4B in 8GB and another in 4GB and both are reproducible.

Tinkerer555, 2023-08-26:
I should add that eeprom for both Pi's have been upgraded to current stable.

Tinkerer555, 2023-08-26:
I have reproduced using:

1 - install Raspberry Pi 64 with desktop from imager
2 - Do not update any packages
3 - install Orca from app menu > Preferences > Recommended Software
4 - reboot
5 - imager will now be broken and unable to navigate to a custom image

And I have tested:

1 - install Raspberry Pi 64 with desktop from imager
2 - Do not update any packages
3 - install all packages except Orca from app menu > Preferences > Recommended Software 
4 - reboot
5 - imager still functions normally

As well as:

1 - install Raspberry Pi 64 with desktop from imager
2 - Update all packages (sudo[apt update][apt full-upgrade][apt clean][reboot])
3 - install all packages except Orca from app menu > Preferences > Recommended Software
4 - reboot
5 - imager still functions normally

I will next test 32 bit Raspberry Pi OS and update.

Tinkerer555, 2023-08-27:
I can now confirm RasPiOS 32 Desktop and RasPiOS 32 Desktop full have the same issue if Orca is installed.

maxnet, 2023-08-27:
Can confirm that it crashes, but not in Imager's own code, but Qt library code.

```
QAccessibleTree::indexFromLogical: invalid index:  -1 0  for  QTreeView(0x6f1d30, name="treeView")
Requested invalid tree cell: -1 0

Thread 1 "rpi-imager" received signal SIGSEGV, Segmentation fault.
0xf2c24068 in ?? () from /lib/arm-linux-gnueabihf/libQt5XcbQpa.so.5
(gdb) bt
#0  0xf2c24068 in ?? () from /lib/arm-linux-gnueabihf/libQt5XcbQpa.so.5
#1  0xf7a247b0 in ?? () from /lib/arm-linux-gnueabihf/libQt5Widgets.so.5
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Would like to test if the issue also still appears in newer Qt library versions -as the one shipped with Bullseye is a couple years old-, before diving into this deeper.
However that will need to until RPI OS moves over to Bookworm.

maxnet, 2023-08-27:
As workaround for the file selection dialog crashing, you do can specify the custom image from a terminal

```
rpi-imager my-image.img
```

Will then be pre-selected in Imager.

spl237, 2023-08-28:
I tried this on the bookworm image and couldn't repeat the problem, so it looks as if it is fixed in the bookworm version of Qt.

tdewey-rpi, 2023-10-19:
Closing on the basis of @spl237's comment. We can't reach into Bullseye to fix a Qt library bug, so the recommendation is to move to Bookworm if you encounter this.

closed 2023-10-19 by tdewey-rpi