raspberrypi/rpi-imager #628: Raspberry pi reported to be 0 bytes size, when it is like 15.5 (16GB)

Meme-Dealer, 2023-08-27:
[image]
I read the size of the raspberry pi, and it is incorrect. How do I get my device to be able to read its size? I have used the rasberry pi imager, which has also failed because of the same reason.

maxnet, 2023-08-27:
Try "Format as FAT32" in Imager.
If that fails as well, double check "controlled folder access" is disabled.

Meme-Dealer, 2023-08-27:
> Try "Format as FAT32" in Imager. If that fails as well, double check "controlled folder access" is disabled.

It is off right now, the format as FAT32 was not working because of the error code "failed to allow dasd on device"

lurch, 2023-08-27:
Maybe a silly question, but are you trying to use a SDHC card in an old card-reader that doesn't support SDHC? (before SDHC, SD cards were limited to 2GB)

Meme-Dealer, 2023-08-27:
No, I am using a micro sd card in a relatively old card reader

Meme-Dealer, 2023-08-27:
Also I should add, 16mb was actually the size of my raspberry pi, the micro sd card was 16 gb

lurch, 2023-08-27:
Yeah, the fact that your screenshot says "MMC/SD USB Device" gave me a hint that it might be a "relatively old card reader", as it has been a long time since I've seen MMC cards mentioned.

micro-SD-card / fullsize-SD-card is just a physical difference, electrically they're the same. So even a micro SD card will be "SDHC" if it's between 4 and 32GB (see https://en.wikipedia.org/wiki/SD_card#Capacity for more info), so if you have an old card reader it **might** not be able to read SD cards / micro SD cards bigger than 2GB.

tdewey-rpi, 2023-10-19:
@Meme-Dealer Did @lurch's suggestion get you any closer to success?

tdewey-rpi, 2024-02-07:
Closing as stale - no update since Oct '23.

closed 2024-02-07 by tdewey-rpi