raspberrypi/rpi-imager #629: Add better UI for adding authorized_keys

CodingSpiderFox, 2023-08-27:
I started using rpi-imager not too long ago.

As I rarely do setup of ssh servers, I don't know what the proper contents of the authorized keys field should look like in rpi-imager.
Currently, the field is only a single line text input which is quite short. It is not clear to me, which parts of my newly generated public ssh public key (ssh-keygen on linux commandline) I need to paste into that field.
The input field should assist me by indicating whether the pasted string looks like a valid public key.

Also, it should be capable of accepting multiple public keys, e.g. when a friend gives me his public key so I can give that person access too while setting up a new raspi installation.

So the input field should also offer a way of adding multiple ssh keys one after the other and also giving me an indication about how many valid keys I already entered. So when I would see a label like "2 valid keys entered", I would immediately see that the  setup seems to be correct in that case.

lurch, 2023-08-27:
See also #326 and #316

maxnet, 2023-08-27:
>As I rarely do setup of ssh servers, I don't know what the proper contents of the authorized keys field should look like in rpi-imager.

If you are using Linux or Mac OS X it will automatically fill in the contents of ~/.ssh/id_rsa.pub and nothing manually should be needed.
Perhaps we should look into doing the same for Windows, but more complicated there as we are not running under normal username but as admin, and user may not be using the official ssh client -which Windows does ship nowadays- but something third-party like Putty...

CodingSpiderFox, 2023-08-27:
I was actually using Linux in that situation. There was nothing I noticed that was telling me the rpi-imager would automatically pick up the id_rsa.pub. I also don't think that this is a good idea because I want to actively decide which keys should be allowed on the raspi and no magic happen in the background that would violate privacy or integrity of the key/the raspberry install without the user actively giving permission to add a certain key to the raspberry

maxnet, 2023-08-27:
>There was nothing I noticed that was telling me the rpi-imager would automatically pick up the id_rsa.pub. 

It will prefill the contents of ssh key box, but you will still need to click "enable ssh" -> "allow public key only" boxes...

Blueshawk, 2024-11-03:
You should not assume the key in .ssh is appropriate. For starters, those are often bundled with the system and so are widely open to abuse and should be replaced. Also, many of us use more than one set of keys for various systems and have them named and in directories. Overwriting keys already in system is also problematic.
  
    A good way to fix this would be to have rpi-imager run ssh_keygen and place the private key in the new system, but place your pub key in the home dir or giving it a name like "hostname.ecdsa" in ~/.ssh or a line in the setup. indicating such in a dialog. Including a quick instruction like $ssh [IPaddress] -i ~/.ssh/hostname.ecdsa" in the setup dialog would help folks as well.
      
  Also worth noting rsa is now sus and not default in ssh.

ngist, 2025-09-16:
So I just updated and noticed that you have to hover over the key and scroll with the mouse wheel to see all the keys?! This is referenced in #1185 step 6 so it seems like it's been reported but I couldn't find a relevant issue. It is not obvious at all that there are multiple keys there. I went to add my other keys and then realized several more were already populated some desired some not.

Here's a screenshot, it looks like there is one key but there are actually 4?!

[image]

closed 2025-12-12 by tdewey-rpi