raspberrypi/rpi-imager #630: Advanced options: save button not working

rbrtmrtn, 2023-09-04:
Thank you for this awesome tool which has saved me a lot of hassle over the years!

I'm using v1.7.5 on macOS Ventura 13.4.1 and am trying to set some advanced options. When I open the dialog and change a few things and hit save, nothing happens. The button flashes so it seems to be registering a click, but the dialog doesn't or give me any confirmation the save worked. If I close the settings and reopen them, any changes I make are not persisting.

Is this a known issue? Does anyone have any suggestions for further troubleshooting?

maxnet, 2023-09-04:
Scroll up and down and see if any input field is highlighted red.
Can happen if you say enable SSH with password authentication, but forget to specify password.

rbrtmrtn, 2023-09-04:
Ah, thank you @maxnet ! That was it 😄 

I think it would be helpful to make the error visible at the same time as the Save button. I'll close this issue and open another to suggest a UX tweak.

Thanks again!

closed 2023-09-04 by rbrtmrtn