raspberrypi/rpi-imager #634: Verification fails using USB3 port

TheMariday, 2023-09-18:
There doesn't seem to be a ticket for this so thought I'd put one in place linking some recent comments on #626 and #41

I've had the same issue where the etcher consistently fails to verify whilst writing to an SD card connected via a card reader plugged into a USB3 interface.

USB2.0 still works fine.

Flash OS: Windows 10
image: Bullseye 64 lite
SD: 8gb Sandisk

Hope this helps and I'll link any more tickets I can find, I imagine that this is going to cause more issues as more devices and laptops move over to USB3.0

tdewey-rpi, 2023-10-20:
I've been using a USB-3 card reader on a laptop, along with a PCI-E card reader built in to my laptop, and both have written and verified across all available ports without issue. FWIW, the laptop is a Dell XPS 15.

As this is also affecting Belena, and I'm unable to reproduce it, I'm minded to suggest this is _not_ in fact an issue with rpi-imager, but is an issue with the hardware it's running on. I'll hold this open until November 15th to allow other users to chime in - but right now I can't see what we can reasonably do to address this.

ig-sinicyn, 2023-11-21:
Hi!
Same issue here. Rpi Imager v1.8.1, Windows 11 (RU locale).

I've checked with three SD cards, three card readers and two PCs - it stable fails when USB3 port is being used.

Other tools such as Rufus and sd imager works without any issue.

tdewey-rpi, 2023-11-23:
Thanks for the report, @ig-sinicyn.

Your point on card speed is particularly interesting - I've been testing with an [A1, U1, C10] card, and see no sign of failure. Unfortunately, as this is a rare occurence and happening in other projects too, I'm inclined to suggest this is not an issue with Raspberry Pi Imager, but instead is possibly a multi-factor problem exposed with individual user's PC and sdcard combinations - both of which are beyond our ability to try and address without being able to reproduce the failure.

In my previous comment, I suggested to close this issue on November 15th. Unfortunately we haven't made any traction in terms of finding a reproducible case, and as such I'm closing this as _Won't Address_ on the basis that I can't see how we can meaningfully make progress on it.

closed 2023-11-23 by tdewey-rpi

ghiathkamel, 2024-02-08:
i have same issue i tried two laptops, many RPI OS
[image]