raspberrypi/rpi-imager #636: CMake Error at CMakeLists.txt:8 (project):   No CMAKE_CXX_COMPILER could be found.

40476, 2023-09-24:
bash screenshot included
```_40476@localhost:/run/media/_40476/a6eab8ba-3e4a-47a9-a991-fb03e3dc07ab/APPS/rpi-imager-qml/build> cmake ../src
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:8 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:8 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
_40476@localhost:/run/media/_40476/a6eab8ba-3e4a-47a9-a991-fb03e3dc07ab/APPS/rpi-imager-qml/build> 
```

maxnet, 2023-09-25:
Missing some details here, like under which Linux distribution you are trying to compile and what you did to get the build requirements...

It complains you do not have a C++ compiler installed.
That typically gets pulled in when you install build-essential on Debian based systems.

40476, 2023-09-28:
my apologies, I have realized and corrected my error. Thank you for your help

closed 2023-09-28 by maxnet