raspberrypi/rpi-imager #637: PiOS bookworm issues

XECDesign, 2023-09-25:
In Bookworm, the mountpoint for the firmware partition has changed from /boot to /boot/firmware.
Also, the regulatory domain is changed through cmdline.txt.

rpi-imager has `/boot` [hard-coded](https://github.com/raspberrypi/rpi-imager/blob/1411ba39b63ad13721629740650c5b31f4e4796f/src/downloadthread.cpp#L972) into it in a [few places](https://github.com/raspberrypi/rpi-imager/blob/1411ba39b63ad13721629740650c5b31f4e4796f/src/OptionsPopup.qml#L787) and also [strips cmdline.txt of additional parameters](https://github.com/raspberrypi/rpi-imager/blob/1411ba39b63ad13721629740650c5b31f4e4796f/src/OptionsPopup.qml#L788).

I'm currently working around this by patching firstrun.sh and cmdline.txt in initramfs, but that's not ideal. And, it's likely to break whenever `firstrun.sh` changes.

maxnet, 2023-09-25:
Just workaround it by having a permanent symlink in RPI OS?

ln -sf /boot/firmware/firstrun.sh /boot/firstrun.sh

maxnet, 2023-09-25:
Still no chance of your scripts parsing cloud-init config instead?
As that is something existing Imager versions can also generate...
Any change done in Imager to figure out whether the image uses /boot or /boot/firmware will naturually require users to upgrade first.

maxnet, 2023-09-25:
And not sure how Imager would need to figure out which path to use if user provided a custom .IMG file instead.

Imager does can read from FAT32 partitions, but doesn't know anything about EXT4 so cannot easily look at /etc/fstab.
Perhaps also put RPI OS generation, instead of just "Raspberry Pi reference 2023-09-22" in issue.txt on FAT partition?

closed 2023-09-26 by XECDesign