raspberrypi/rpi-imager #646: Please Add To RPI Imager Configuration Settings Ability to Enable Serial Console!

Jibun-no-Kage, 2023-10-06:
Please Add To RPI Imager Configuration Settings Ability to Enable Serial Console!  This would be a great addition to the imager settings options.

maxnet, 2023-10-06:
Not sure if there's much demand for that. Would expect it to be on by default on Pi 5 now it has separate port.

Jibun-no-Kage, 2023-10-06:
If you look at all the embedded Pi devices, that alone would suggest that adding ability to enable serial console support via the imager is long over due.  Every Pi I have, and I have about 20, only one is connected to HDMI, for HDMI-CEC, all are headless or serial connected.  Even the remote sensor solutions I have via Pi devices I have a serial IO board connected and I just walk up and connect to them via PuTTY via serial.  I can use my phone, laptop, etc. to validate status and function as needed if network is down or not available.  There are countless examples on the internet of Pi devices using serial.  IMHO, the Pi 5 design acknowledges how often direct serial is used over HDMI, never mind embedded compute modules being all but serial oriented, via expansion boards.

timg236, 2023-10-07:
On Pi5 BOOT_UART=1 in the EEPROM config but that will not automatically set enable_uart=1 on config.txt.

It might be nice to be able to append a user-defined chunk of text to config.txt via the imager.

```
[all]
[pi5]
enable_uart=1
usb_max_current_enable=1
dtparam=pciex1
```

N.B. It's possible to do something similar by adding a `[config.txt]` section the EEPROM config but I guess it depends on whether you are using multiple images on a single boards or flashing multiple images for multiple boards as to whether the EEPROM config settings extension is useful.

timg236, 2023-10-09:
> > On Pi5 BOOT_UART=1 in the EEPROM config but that will not automatically set enable_uart=1 on config.txt.
> 
> What is the logic behind that?

BOOT_UART has the same behaviour as on previous models. Maybe in the future someone will add support for automatically enabling the OS console and plumbing through UART baud rate (which can be > 115200) from boot-onwards.