raspberrypi/rpi-imager #647: Update Home Assistant main entry to use new logo

agners, 2023-10-09:
Home Assistant recently became a new logo (see [this blog post](https://www.home-assistant.io/blog/2023/09/17/a-refreshed-logo-for-home-assistant/)). Subitem json got updated yesterday (see https://github.com/home-assistant/version/pull/325). However, the main entry needs an update as well. So the new entry should use the following image URL:

```
"icon": "https://version.home-assistant.io/brand/logo.png",
```

maxnet, 2023-10-09:
Changing OS repo files is not my department. But do am curious why a new logo needs a new URL, when your old icon URL is already rpi-specific?

lurch, 2023-10-09:
ping @XECDesign

agners, 2023-10-09:
> Changing OS repo files is not my department. But do am curious why a new logo needs a new URL, when your old icon URL is already rpi-specific?

Honestly, I forgot that we use that URL here separately :see_no_evil: 

But I'd prefer to change the URL here now that the URL already got changed on our end.

XECDesign, 2023-10-09:
Done

agners, 2023-10-09:
Thanks, LGTM!

closed 2023-10-09 by agners