raspberrypi/rpi-imager #649: Advanced options window walks across the screen when adjusting height

cillian64, 2023-10-09:
When you resize the height of the new advanced options dialog it walks across the screen.

System: GNOME, Wayland, Debian Bookworm
[Screencast from 2023-10-09 12-00-51.webm](https://github.com/raspberrypi/rpi-imager/assets/1373016/698aa5de-99f2-4061-8af3-7bc7449c43a6)

maxnet, 2023-10-09:
Not reproducible on Kubuntu (KDE). 
So may be limited to GNOME/Wayland.

cillian64, 2023-10-10:
I don't see this in wayfire on my Pi, so it seems like it might be GNOME-specific.

If I'm the only person seeing this then I can have a look into it and see if I can figure out what's going on.

maxnet, 2023-10-11:
>If we make the advanced options dialog fully resizeable then it will resolve this bug

I was indeed too lazy to add the logic to expand the text fields as well, when the width is adjusted. ;)
And test that on a dozen different platforms and windowing systems.
So yes, I made the width fixed.

If someone else does want to spend time on that, be my guest. :)