raspberrypi/rpi-imager #650: Advanced options selection doesn't work in text box

ghollingworth, 2023-10-09:
Not sure why, but if you double-click on one of the text entries it doesn't select the text.

For example, open the advanced options, click the "Set hostname" tickbox and then double-click "raspberrypi" that should select the whole word but doesn't.  Also click and drag doesn't work.

maxnet, 2023-10-09:
Do are running a build more recent than: https://github.com/raspberrypi/rpi-imager/commit/ce549cadb231614ffc544e538c5e873b75217bb6 ?

Having "selectByMouse: true" on the fields is important.

tdewey-rpi, 2023-10-13:
@ghollingworth This should be addressed in the next release, which is pending on #645 and the associated translation changes.

tdewey-rpi, 2023-10-19:
Resolved in v1.8.1. Try it out!

closed 2023-10-19 by tdewey-rpi