raspberrypi/rpi-imager #653: Worked once but from then on "no kernel"

hagenbuch, 2023-10-10:
I am using the imager 1.7.5 on Linux Mint.

I could flash an OS Lite on my 8 GB Micro SDHC (including setting a user, time and keyboard) on a new Pi 3B only once but then, on a different SD card, after a while, the green LED gives me 7 pulses which means there is no kernel.

And yes, on rootfs partition, /boot is empty.

On boofs partition, there are files like kernel.img, kernel7.img, kernel7l.img, kernel8.img but the config.txt does not even contain the word kernel. File fristrun.sh is still there, I see references to the first user "rpadmin" I have to created but nothing else i can understand.

Any help welcome!

hagenbuch, 2023-10-11:
I think I found the solution:

When booting for the first time, Raspbian seems to wait for network boot even if it has a valid system on SD flash.

On desktop systems, it may display this fact on screen but on a "headless" system (Lite version) we don't get to see it.

So: DO NOT CONNECT TO LAN when booting for the first time, leave it at least five minutes to get housekeeping done before connecting LAN (or WLAN).

closed 2023-10-11 by hagenbuch

lurch, 2023-10-11:
> When booting for the first time, Raspbian seems to wait for network boot even if it has a valid system on SD flash.

Sounds like you must have changed your boot-priority options in the Pi4 EEPROM.

hagenbuch, 2023-10-11:
Nope, I used it just as I bought it. I do not intend to hook up a screen in my use case.

maxnet, 2023-10-11:
>When booting for the first time, Raspbian seems to wait for network boot

How did you come up to that conclusion without screen?

hagenbuch, 2023-10-11:
"seems" is the key here. Actually I have spent about a day to struggle with the installation of four identical new raspberry Pi 3 B here. The all work fine with the desktop version but the "Lite"...  seems the imager suddenly got "bookworm" version, so some started as bullseye, some as bookworm, two not at all. 

I had to connect them all to a screen to see what's going on and when they stalled (multicolored screen, LED blinking..).

In almost all cases, the blues screen "Making SSH key" or the like appeared, some screens said they would reboot in five seconds, some just booted into nothing, some booted.

Its been quite a nightmare so far.

Tomorrow, I will check another new SDcard from scratch, double check everything and report here.

lurch, 2023-10-11:
> seems the imager suddenly got "bookworm" version, so some started as bullseye, some as bookworm

Yes, Raspberry Pi OS Bookworm was [released today](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/), and RPi Imager will always automatically offer you the "newest" OS. However you can still use RPi Imager to flash Bullseye to a card by choosing "Raspberry Pi OS (other)" and then scrolling down to one of the "Legacy OS" options.

hagenbuch, 2023-10-12:
Thank you, I'll try!