raspberrypi/rpi-imager #655: Feature : Find Raspberry Pi SSH without knowing the IP

mats-nk, 2023-10-11:
Today do the rpi-imager configuring mDNS/Avahi as a workstation and that is god, but I would like to take it a bit further.

When SSH is checked it would be an enhancement if it also announced via Avahi that there is a SSH service:

This is quite simple, just add a new file, ex.: `/etc/avahi/services/ssh.service` With the content:

This would enable developers of ssh clients to "discover" SSH servers on the local network and isn't this in the spirit of "Zero network configuration"?

```
<service-group>
  <name replace-wildcards="yes">SSH server on %h</name>

  <service>
    <type>_ssh._tcp</type>
    <port>22</port>
  </service>
</service-group>
```

mats-nk, 2023-10-12:
Thx for your reply @lurch.
Where do the OS pick up that the ssh checkbox has been checked in imager?
And how do I ping someone, is it by @XECDesign ?

And I agree this is a feature of the OS.

XECDesign, 2023-10-12:
It should be opened here: https://github.com/RPi-Distro/repo/issues

Won't be able to take a look at adding this feature any time soon though.

tdewey-rpi, 2023-10-13:
Thanks for the report, @mats-nk. As you've been redirected to the correct project, I'm going to close this issue. If you feel that we've got something directly related to fix in the imager, please leave another comment and I'll take a look.

closed 2023-10-13 by tdewey-rpi