raspberrypi/rpi-imager #659: Apple developer subscription question

tmxkn1, 2023-10-14:
I would like to contribute to bug fixes on macOS. Do I need to pay a 99 USD annual subscription to be able to work on the project?

maxnet, 2023-10-14:
No subscription is necessary to build and run Imager on your own computer.
Only the person that builds and releases Imager for distribution to others need to pay the Apple tax.

tmxkn1, 2023-10-14:
Thanks for clarification 

> No subscription is necessary to build and run Imager on your own computer. Only the person that builds and releases Imager for distribution to others needs to pay the Apple tax (currently me).
> 
> Builds that are not uploaded to and notarized (counter-signed) by Apple will not run on other people's computers in a normal manner. (without requiring users to override Apple's security).

closed 2023-10-14 by tmxkn1