raspberrypi/rpi-imager #660: macOS build instructions

rejas, 2023-10-14:
Tried the new macOS build instructions that are in preparation here: https://github.com/tdewey-rpi/rpi-imager/tree/dev/macos/use-brew

Compiling results in an error though:

```
23:41:49: Führe Schritte für Projekt rpi-imager aus...
23:41:49: Starte: "/opt/homebrew/Cellar/cmake/3.27.7/bin/cmake" --build /Users/michael/Projects/rpi-imager/build-src-Desktop_Qt_5_15_2_clang_64bit-Debug --target all
[ 13%] Built target libzstd_static
[ 37%] Built target cmliblzma
[ 82%] Built target archive_static
[ 84%] Automatic MOC for target rpi-imager
[ 84%] Built target rpi-imager_autogen
[ 85%] Generating rpi-imager_de.qm
Updating '/Users/michael/Projects/rpi-imager/build-src-Desktop_Qt_5_15_2_clang_64bit-Debug/rpi-imager_de.qm'...
    133 Übersetzungen wurden erzeugt (132 abgeschlossen und 1 nicht abgeschlossen)
    6 nicht übersetzte Texte wurden ignoriert
[ 85%] Automatic RCC for translations.qrc
[ 85%] Building CXX object CMakeFiles/rpi-imager.dir/rpi-imager_autogen/EJRQKI7XPS/qrc_translations.cpp.o
[ 85%] Linking CXX executable rpi-imager.app/Contents/MacOS/rpi-imager
/bin/sh: /Users/michael/Qt/5.15.2/clang_64/bin/macdeployqt: No such file or directory
make[2]: *** [rpi-imager.app/Contents/MacOS/rpi-imager] Error 127
make[2]: *** Deleting file `rpi-imager.app/Contents/MacOS/rpi-imager'
make[1]: *** [CMakeFiles/rpi-imager.dir/all] Error 2
make: *** [all] Error 2
```

Qt5 got installed via brew into /opt/homebrew/opt/qt@5 though...

maxnet, 2023-10-14:
Double check the build configuration selected in Qt Creator.
It scans your system for all Qt SDK installations present.
And 5.15.2 may be an old one.

tdewey-rpi, 2023-10-16:
That's extremely weird - the key line is:

`/bin/sh: /Users/michael/Qt/5.15.2/clang_64/bin/macdeployqt: No such file or directory`

This path appears to have come from a find_program directive in the root CMakeLists:

`    find_program(MACDEPLOYQT "macdeployqt" PATHS "${${QT}_DIR}/../../../bin")
    if (NOT MACDEPLOYQT)
        message(FATAL_ERROR "Unable to locate macdeployqt")
    endif()`

So it looks suspiciously like your configured QT path is incorrect - as it would in theory be looking in: `/opt/bin/`, which also doesn't look too healthy.

I _suspect_, but cannot confirm without a macOS install, that we need to adjust that search path to reflect Homebrew's paths regardless, and I'll not push for that PR to get merged until I can fully replicate the setup.

maxnet, 2023-10-16:
>So it looks suspiciously like your configured QT path is incorrect - as it would in theory be looking in: /opt/bin/, which also doesn't look too healthy.

Not sure if it is incorrect.
Qt creator will auto-detect and offer you to use ANY Qt version installed on your machine.
Which can be more than the one in /opt/bin
Think Michael has previously installed 5.15.2 as well...

The first time you open a project it will show a dialog like this asking which ones you actually want to use:

[image]

Between the ones you checked, you can then switch at the bottom left of screen:

[image]

(If you have problems navigating the Qt creator IDE, you can also just delete src/CMakeLists.txt.user manually, in which case you will be prompted again for Qt version/toolchain next time you open the project.)

tdewey-rpi, 2023-10-19:
Thanks for trying the instructions out, @rejas. Unfortunately, because Homebrew is unable to provide a convenient way to manage dependencies for both architectures on one machine, I've been forced to halt work on those updated instructions. When we get past the Pi5 shipping date, I will consider looking at the existing instructions, to see if there are any steps we can make simpler.

tdewey-rpi, 2024-02-07:
Closing this one out - there's not a great way forward re: build instructions - but I'll continue passively looking at opportunities to simplify this.

closed 2024-02-07 by tdewey-rpi