raspberrypi/rpi-imager #661: Imager on macOS Mojave is reporting expired certificate for downloads.raspberrypi.com

trejan, 2023-10-15:
Had a few reports on the forum about this issue. Raspberry Pi Imager running under Windows works. I can't confirm the bug as I'm not a Mac user but I have checked that the certs are valid.

See https://forums.raspberrypi.com/viewtopic.php?t=357672 and https://forums.raspberrypi.com/viewtopic.php?p=2145006

maxnet, 2023-10-15:
Problem may be that the webserver is sending a LetsEncrypt intermediate certificate for downloads.raspberrypi**.com** that is cross-signed by the DST X3 root certificate that has expired.

Problem may go away if we would switch to the self-signed ISRG Root X1 intermediate certificate from: https://letsencrypt.org/certificates/
But that's up to the server folks (Mythic?)

rin67630, 2023-10-16:
on macOS Mojave, the solution is here:
https://macresearch.org/dst-root-ca-x3-expired-mac/

But, anyhow, the error message on RPI imager **must** mention **which** certifcate is not valid, allowing the user to fix the problem on the source.

maxnet, 2023-10-16:
>on macOS Mojave, the solution is here

Deleting the root certificate on the client can indeed also work.
Not mentioning that certificate server-side (by using the self-signed ISRG Root X1) should also work, and does not require client changes.

>But, anyhow, the error message on RPI imager must mention which certifcate is not valid

Not practical to implement.
Outsources downloads to libcurl, which in turn outsources TLS to various OS specific libraries (openssl or gnutls depending on Linux distribution, yet different system libraries for Windows and Mac OS)
We just pass through the error message received from that, and never get to see the certificate chain ourselves.

mudge, 2023-10-16:
We currently use the default "long" chain handed out by Let's Encrypt but there's a trade-off in compatibility with older software, see https://community.letsencrypt.org/t/long-default-and-short-alternate-certificate-chains-explained/162526

Either way, some clients are going to be incompatible. I raised this with Mythic Beasts but, while switching to the "short" chain may fix this specific error, they are concerned it will introduce more compatibility problems than it fixes.

ghollingworth, 2023-10-16:
Could print out the URL that failed, at least then the user can try the URL in a browser to find out what the issue is?

lurch, 2023-10-16:
Wikipedia https://en.wikipedia.org/wiki/MacOS_Mojave says "Unsupported as of October 2021".

maxnet, 2023-10-16:
>We currently use the default "long" chain handed out by Let's Encrypt but there's a trade-off in compatibility with older 
>software, see https://community.letsencrypt.org/t/long-default-and-short-alternate-certificate-chains-explained/162526

Do ancient Android phones have any business on https://downloads.raspberrypi.com/ ?

maxnet, 2023-10-16:
Alternatively can we change the RPI OS downloads in os_list.json to use downloads.raspberrypi.org instead of .com ?

.org has a proper Sectigo certificate instead of LetsEncrypt.

maxnet, 2023-10-16:
>Wikipedia https://en.wikipedia.org/wiki/MacOS_Mojave says "Unsupported as of October 2021".

Could be.
But market share of OSX users that run something older than Catalina/Big Sur/Monterey is still 8%
https://gs.statcounter.com/os-version-market-share/macos/desktop/worldwide

circlinglights, 2023-10-18:
The error is still occurring for me. I installed the keychain and restarted but no change. Can it not be fixed?

[image]
[image]
[image]

maxnet, 2023-10-18:
Likely need to REMOVE the expired DST certificate from your system, to trick the system. Then it doesn't know the root certificate that is referenced to is expired.

Maybe easier to replace the entire file it is in.
As in download to your home directory: https://curl.se/ca/cacert.pem

Start a terminal:

```
sudo mv /etc/cert/ssl.pem /etc/cert/ssl.backup
sudo mv cacert.pem /etc/cert/ssl.pem
```

circlinglights, 2023-10-18:
I learned in the other thread that I can just download the image itself and select that from the Imager program, and I no longer get the error that way. It's still a workaround but I can get by with it. Thanks, all.

nezort11, 2023-12-08:
**1. Download image from the website**
[image]

**2. Use custom image to write**
[image]

tdewey-rpi, 2023-12-15:
Closing this issue:

- ISRG Root X1 is valid through 2035, R3 is valid through 2025.
- Mojave is no longer supported by Raspberry Pi Imager (we require 10.15/Catalina at the earliest, and I strongly encourage users to patch their desktop OS').
- There are workarounds for configurations that are not supported in either current macOS or Raspberry Pi Imager

closed 2023-12-15 by tdewey-rpi