raspberrypi/rpi-imager #671: Embedded imager kernel panic

cillian64, 2023-10-17:
Steps to reproduce:
- Build the embedded imager from latest commit on `qml` which is `f28b9d0`
- Copy the files to a FAT-formatted USB drive
- Boot imager from the USB drive on a Pi 4
- Then insert a microSD card
- In embedded imager:
  - Select an OS
  - Select the microSD card to image to
  - Click Next
  - Click Edit Settings

The result is a kernel panic `Attempted to kill init!`

maxnet, 2023-10-17:
Apply this patch manually to Imager source: https://github.com/raspberrypi/rpi-imager/blob/qml/embedded/imager/package/rpi-imager/0001-rpi-imager-change-window-to-popup.patch
(Applying it automatically does not seem to work properly with SITE_METHOD=local)

Or have a go at using Wayland instead of EGLFS, if you do want to have multiple windows instead of a single one.
But that will probably enlarge image size by quite a bit.

maxnet, 2023-10-17:
Oh, and "kernel panic" is a bit over stated. "Attempted to kill init" is just that the main program (init) has ended.
(Which it did because it cannot create extra top level windows in EGLFS mode, that is meant for full screen applications)

cillian64, 2023-10-17:
Thanks, I saw that commit but didn't realise it needed manually applying.  I'm overdue for learning about buildroot anyway so I'll see if I can figure out how to make it automatically apply.

> Oh, and "kernel panic" is a bit over stated. "Attempted to kill init" is just that the main program (init) has ended.

Right, that makes sense. I've not seen this before so took it literally.

maxnet, 2023-10-17:
>Thanks, I saw that commit but didn't realise it needed manually applying. 

Normally *.patch in a buildroot's package folder is applied automatically during download & extract.
But we currently not downloading anything, so it is skipping that phase.

Either uncomment the github line here and remove the path and METHOD=local lines: https://github.com/raspberrypi/rpi-imager/blob/qml/embedded/imager/package/rpi-imager/rpi-imager.mk#L9
Or create a PRE_CONFIGURE_HOOKS function that runs patch manually.

PR #674 (2023-10-18): Embedded: Automatically apply patches
When using local site, buildroot does not automatically apply patches. The buildroot manual recommends using the POST_RSYNC hook to apply patches if needed, so implement this to automatically apply patches.

Resolves #671
commit 3a20f1cc (2023-10-18): Embedded: Automatically apply patches

When using local site, buildroot does not automatically apply patches.
The buildroot manual recommends using the POST_RSYNC hook to apply
patches if needed, so implement this to automatically apply patches.

Resolves #671
closed 2023-10-18 by tdewey-rpi