raspberrypi/rpi-imager #690: Raspberry Pi OS Bookworm absent for Raspberry Pi's 3 and older

grandsilvergalaxy, 2023-10-19:
No Raspberry Pi OS on top
[image]
Raspberry Pi OS (Other Category) No Bookworm
[image]

tdewey-rpi, 2023-10-20:
Thanks for the report, @grandsilvergalaxy.

We're not explicitly recommending Bookworm for Pi3 and older at this time. If you want to write Bookworm, I'd advise you use the 'No filtering' option in the device selector.

Closing, as this is working as intended at this time.

closed 2023-10-20 by tdewey-rpi

b-reich, 2023-11-04:
[image]

this is really confusing
its not listed on the website that it is not recommended
https://www.raspberrypi.com/software/operating-systems/

b-reich, 2023-11-04:
> Thanks for the report, @grandsilvergalaxy.
> 
> We're not explicitly recommending Bookworm for Pi3 and older at this time. If you want to write Bookworm, I'd advise you use the 'No filtering' option in the device selector.
> 
> Closing, as this is working as intended at this time.

[image]

this is really confusing
its not listed on the website that it is not recommended. it is even recommended
https://www.raspberrypi.com/software/operating-systems/